hneemann/Digital #3: "BCD-Zähler" zum Hochzählen von BCD basierend auf einer Taktleitung

M-Reimer, 2017-02-11:
Zum Simulieren einer Schaltung benötige ich einen Zähler der mindestens 4 Bit BCD hochzählen kann (ausgehend von einer Signalquelle).
Ich habe das behelfsmäßig mit Flip-Flops umgesetzt, es wäre aber hilfreich einen entsprechenden Baustein direkt nutzen zu können.

M-Reimer, 2017-02-12:
Das ist nicht genau das, was in unserer konkreten Aufgabe gefordert wurde.
Aufzubauen war eine Digitalschaltung die basierend auf 4 Bits ein 7-Segment-Display ansteuert.
Also klassiche Binär-Codierung

0000 -> 0
0001 -> 1
0010 -> 2
...

Im zweiten Schritt sollte die 7-Segment-Anzeige dann automatisch hochzählen. Mit etwas Googlen haben wir geschafft einen passenden Zähler mit Flipflops und ein paar Und-Gliedern zu bauen.

Der Aufgabentext sagt:
Tipp: Mit einem Rechteckgenerator (Takt) und einem Zählerbaustein ist dies recht einfach zu realisieren.

Eben diesen "Zählerbaustein" haben wir entweder nicht gefunden oder er fehlt.

hneemann, 2017-02-12:
Es gibt eine Siebensegmentanzeige, welche einen entsprechenden 4-Bit Eingang hat. Diese hat den entsprechenden Decoder quasi eingebaut. In dem Beispiel welches oben Angehängt ist, finden Sie auch den entsprechenden Zählerbaustein. Wenn Sie BCD zählen wollen, müssen Sie den Zähler nur bei erreichen der 10 zurücksetzen. Evtl. schauen Sie sich das oben angehängte Beispiel noch einmal genauer an. Oder liegt Ihr Problem im Aufsplitten der 4-Bit Leitung in 4 Leitungen mit jeweils einem Bit? Dafür können Sie den Splitter verwenden.

M-Reimer, 2017-02-12:
Genau das war das Problem! Es ist zumindest für mich als "Digitaltechnik-Anfänger" nicht auf Anhieb schlüssig, dass an einem Ausgang mehrere "Bits" rausfallen. Mit einem Splitter habe ich dann die gewünschte Konfiguration "ein Bit pro Ausgang" herstellen können.
Ich habe jetzt zwei 7-Segmentanzeigen parallel (die mit Eingang pro LED an meinem selbstgebauten Decoder und die mit Decoder direkt nach dem Zähler). Funktioniert jetzt einwandfrei.
Danke für die Unterstützung!

closed 2017-02-12 by M-Reimer