hneemann/Digital #5: Backspace wie Entfernen-Taste behandeln - zum Löschen von Elementen

carstenhag, 2017-04-25:
Hallo,

das Problem betrifft hauptsächlich Macs, da man `fn + backspace` drücken muss um ein entfernen zu betätigen, aber auch auf Windows-Laptops würde dies die Usability verbessern.

hneemann, 2017-04-25:
Ich habe gerade keinen Mac zur Hand. Würden Sie kurz testen, ob von HEAD auf dem Mac die Backspace-Taste erkannt wird?

carstenhag, 2017-04-25:
@hneemann Mit dem Commit wird die Taste korrekt erkannt und die Elemente werden gelöscht, vielen Dank!

hneemann, 2017-04-25:
added in v0.12

closed 2017-04-25 by hneemann

hneemann, 2017-06-08:
@carstenhag: Konnten Sie inzwischen v0.11.1 und v0.12.1 bzgl. der Schriftgröße vergleichen? Ich habe die Befürchtung, dass auf dem Mac mit Retina-Display die Schrift im Menü in v0.12.1 zu groß sein könnte.