hneemann/Digital #6: High DPI support could be improved

guiweber, 2017-05-08:
On a high DPI screen, the text, icons and buttons are very small and hard to read.

Here is a screenshot of the settings window. The font size used for the title is scaled properly and is easy to read. However everything inside the window is much smaller than it should be and hard to read.

[image]

hneemann, 2017-05-08:
Can you try the [assembler](https://github.com/hneemann/Assembler/releases/latest) application? How does it look like on your display?

guiweber, 2017-05-08:
The text in the assembler application is indeed more legible, but still small. However, the icons and the scroll bar are the same size and would benefit from twice as large.

Here is a side by side screenshot:
[image]

Also, here is a side by side screenshot with the Arduino IDE where the text and icons and controls are, in my opinion, scaled to the perfect size on my screen:
[image]

hneemann, 2017-05-09:
Could you please try the current digital head revision? How does it look like on your display?

guiweber, 2017-05-10:
Sure, but would you mind providing an exe or jar file?

hneemann, 2017-05-11:
You can find a pre realase version [here](https://drive.google.com/file/d/0B1nBRhIo-nQuUWRqYmRLRURBWmM/view?usp=sharing).

guiweber, 2017-05-11:
The font size and icons are now the perfect size. There are still a few minor issues where some items are too small:
- The Order Input, Order Output and Order Mesurement Values are too small
[image]

- The shortcuts in the edit menu are too small:
[image]

- The component menu tab could be wider (Arithmetic barely fits in). There are other menus like this but this is the most user facing:
[image]

- In the Help, the component graphics are too small and it is hard to see the input and outputs. Here is the Tree View (legible) compared to the Help (too small):
[image]

- Filesystem dialogs could be taller by default. They only show one row of file/folders when the dialog opens:
[image]

hneemann, 2017-05-11:
I have fixed the most issues you mentioned. Thank you for your feedback.
By the way: Which kind of screen you are using? It seems to me that it has far more then 200dpi.

guiweber, 2017-05-11:
Thanks!

My screen is a 15 inches laptop screen with a 4k resolution, which brings it close to 300dpi.

By the way, is there a library for Digital that provides the 74xx series TTL? I was checking out Digital as a candidate to replace Logisim for a logical circuit class but we need to have the 74xx series available out of the box.

hneemann, 2017-05-11:
Are you interested in the logic functions only or in something looking like a 74xx dil IC on the screen, including the real pin positions and so on?

guiweber, 2017-05-12:
I'm looking for something with the real pin positions as this really help the students when they get to assemble their physical circuit afterward.

hneemann, 2017-05-12:
Did you think of something like that?
[image]

guiweber, 2017-05-12:
Yes, that is exactly what I am looking for!

hneemann, 2017-05-12:
I have added a DIL package shape which can be used to represent an embedded circuit.
But there are a whole bunch of available circuits and I do not have time to create all available ICs. And I also do not need this myself, because I use a CPLD in my lab exercises.
Which 74xx-ICs do you need for your lab exercises?

hneemann, 2017-05-12:
[Here](https://drive.google.com/file/d/0B1nBRhIo-nQuUWZ5cGphdWItQVU/view?usp=sharing) is a pre release version including the new DIL features. If you like, you can play around with the new examples.

guiweber, 2017-05-12:
Thanks, I'll let you know when I have a chance to take a look, probably next week. Cheers!

closed 2017-05-12 by hneemann

guiweber, 2017-05-25:
Hi @hneemann

I tested the DIL feature and it would work very well for our needs. Most of the 74xx IC that we need are already there but I could help add the others if any are missing. What is your ETA for a release with this feature?

There is a lot I love in Digital, but there are several usability issues that would still drive me to use Logisim instead. I have noted them below, let me know what you think and if you have questions.
- Drawing wires is a lot more tedious in Digital. In Logisim, wires are always routed at 90 degrees which makes for nicer circuits and makes it easier to draw them. I like the flexibility offered by Digital, but I would suggest making the wire routing snap to 90 degrees by default and offer to draw at any angle only when holding a key (e.g. the ALT key), much like how the snap to grid works in many apps. Also as a result, drawing 90 degree angles currently requires clicking on the schematic to "set" the wire at a position, which makes it more time consuming than in Logisim.
- Selecting and moving single components/wires is a pain in Digital since you can't just click on a single component and move it, because it won't route the wires to the new location. You need use the selection box instead and select the wires with the component. This is compounded by the fact that the routing is not done at 90 degrees by default, so even if you drew your wires with 90 degrees angles, moving a component will change the angle at which the wires are connected. The only solution is to delete the wires and draw them again. Deleting single wire segments is also a pain because clicking on a wire segment does not select it and the selection box makes it practically impossible to select a single segment. 
- Drawing wire does not work well on my 4k HiDPI monitor for some reason. The mouse clicks do not always seem to register properly, making it impossible to work efficiently. My other, almost identical system, with a regular HD monitor works just fine.
- Even on my regular monitor, mouse clicks sometimes don't seem to register when selecting a component in the tree view for placement. Clicking on the connectors of components to attach wires can also be tricky (the click area seems small).
- Right clicking on a selected component doesn't open the component's property window.
- Selecting a component in Logisim shows the properties of the component under the component tree view, which makes it much faster to change the component's properties.
- Short circuits during the simulation generate a pop-up when the short happens, which is very annoying. Logisim handles this in a more graceful and user-friendly manner in my opinion, which is to paint the wire red. Logisim also shows potential short circuits before the simulation start, which I find useful.
- Also, Logisim shows the initial circuit state (colored wires) before the simulation starts, which allows identifying issues while designing the circuit.
- There is no "undo" functionality in Digital.

Finally, I found two minor enhancements that could be added while comparing with logisim:
- A clean background is nice for screenshots, but It would also be nice to have the option to display a point grid in the background to better align components.
- Logism-Evolution seems to use some kind of anti-aliasing which makes the component shapes appear much smoother and nicer than in Digital (see comparative screenshot).
[image]

hneemann, 2017-05-25:
Hi @guiweber,

thank you again for your feedback.

Our lab classes are currently underway and I do not want to release a new version before these are not finished. I plan to release a new version in two weeks.

Here are my remarks about your comments:

- _Drawing wires is a lot more tedious in Digital_ 
This is funny: I had always problems with drawing wires in Logisim and therefore implemented the line drawing in a more convenient way. But i will think about an alternative drawing mode.

- _Selecting and moving single components/wires is a pain in Digital_
Hmm... A left click on a component allows moving it. Drawing a select rectangle around the component moves also the wires connected to the component. A right click with CTRL down on a wire selects that wire which allows to move or delete it. With the select rectangle you can pick any wire edge to move it around.

- _Drawing wire does not work well on my 4k HiDPI monitor for some reason._
This effect is hard to understand because all drawing and clicking actions on the component pane are completely independent of the scaling and resolution. Does this happen on all zoom levels?

- _Even on my regular monitor, mouse clicks sometimes don't seem to register when selecting a component in the tree view for placement._
Hmm.. The tree view is the standard Swing JTree component which is rock solid. And it works just fine on all machines I have tested it. I do not understand what is going on. Which OS and Java version you are running?

- _Clicking on the connectors of components to attach wires can also be tricky (the click area seems small)._
There is no clicking area for drawing wires at all. Wires always snap to the grid, thats it. All connectors on all devices are placed exactly on this grid. Otherwise you would not be able to connect them.

- _Right clicking on a selected component doesn't open the component's property window._
Never? Or only sometimes? I use this all the time and have never observed such a behavior. Does this happen only on the high-DPI screen?

- _Selecting a component in Logisim shows the properties of the component under the component tree view, which makes it much faster to change the component's properties._
I do not see the difference. From my understanding the only difference is the need in Digital to click an Ok button. Which also allows to abort the operation which is not possible in Logisim.

- _Short circuits during the simulation generate a pop-up when the short happens, which is very annoying._
Maybe you are right. I have not seen this dialog for a long time. :-)

- _Also, Logisim shows the initial circuit state (colored wires) before the simulation starts, which allows identifying issues while designing the circuit._
Thats not true. If the simulation does not run in Logisim the colors of the wires are not updated. This only works if you modify the circuit while the simulation runs. Modifying the circuit while the simulation runs is not possible in Digital for good reason: This can cause strange situations, such as a working circuit which does not start again and so on.

- _There is no "undo" functionality in Digital._
Thats a real pain point also in my opinion. To do this right is a lot of work. Therefore this issue is still open.

- _A clean background is nice for screenshots, but It would also be nice to have the option to display a point grid in the background to better align components._
I don't feel the need but I will add a grid. ;-)

- _Logism-Evolution seems to use some kind of anti-aliasing which makes the component shapes appear much smoother and nicer than in Digital_
Thats just a flag which I did not set to improve the drawing speed. I will try to enable anti-aliasing and observe the behavior.

guiweber, 2017-05-25:
Thanks for your answer @hneemann 
I think I have identified a bug that was causing several of the issues I was telling you about. My biggest remaining concern would be with the wire routing system.

_Selecting and moving wires_
- CTRL+Click works well, but I wouldn't have found it if you hadn't tell me!
- I feel that the way it is implemented in Logisim is much more intuitive: simply clicking on a wire selects it. Clicking on a wire and holding the mouse button down will draw a new wire connected to the first wire.

_Selecting and moving components_
- Moving a single components does re-route the wires, which need to be routed manually.
- Moving a component and connected wires often does not give nice results because of the angled wires. In addition, angled wires cannot be straightened and need to be deleted and drawn again
- Here is a screenshot to show how moving a simple gate down means trouble
[image]
- In Logisim, moving a single component gives acceptable results, and moving any wire that is misplaced is easier because it will move the wires connected to it at the same time and not the single wire segment:
[image]

_Drawing wire does not work well on my 4k HiDPI monitor for some reason._
- See issue #7

_Even on my regular monitor, mouse clicks sometimes don't seem to register when selecting a component in the tree view for placement._
- See issue #7

_Clicking on the connectors of components to attach wires can also be tricky (the click area seems small)._
- Seems related to issue #7 as well

_Right clicking on a selected component doesn't open the component's property window._
- No, right clicking just deselects the component even if done while the mouse is over the component. It may be a personal preference, but I would find it more user friendly if it opened the property window. I have a tendency to select the component to open it's properties... but I guess I would get used to how it works.

_Selecting a component in Logisim shows the properties of the component under the component tree view, which makes it much faster to change the component's properties._
- It is not so bad but when you have a lot of components to modify less clicks make things faster.

_Short circuits during the simulation generate a pop-up when the short happens, which is very annoying._
- Haha. Well our students do it a lot at the beginning (e.g. connecting two gate outputs together)

_Also, Logisim shows the initial circuit state (colored wires) before the simulation starts, which allows identifying issues while designing the circuit._
- You are right, the simulation needs to be enabled in Logisim. The difference is that the simulation is enabled by default. Would there be a way in Digital to place the components with the simulation active (initial state only), but automatically reset the simulation when pressing "start"?

hneemann, 2017-05-26:
Hi @guiweber,

- _Selecting and moving wires_
  - _CTRL+Click works well, but I wouldn't have found it if you hadn't told me!_
	Its explained in the FAQ section of the documentation. ;-)
	The documentation is so far very incomplete, but this issue is mentioned. ;-)
   - _I feel that the way it is implemented in Logisim is much more intuitive: simply clicking on a wire selects it._
   From my experience it is never useful to select a single wire segment. This is the reason this function came in so late.
   In most cases you want to move more segments at the same time, which is much easier if you select as much wire segments as needed by the selection rectangle. Even if you select a single edge of a long wire.
   But I think that's mainly a question of habit. Maybe i will add an auto routing feature in the future, but in my opinion that does not give much benefit and is hardly worth the time it takes to implement it.

- _Selecting and moving components_
I decided to allow both: To pick a component to move it away keeping all the connected wires in place, or to pick a component and also move the connect wires.
The result of the latter operation looks different in Logisim and Digital thats right, and maybe the Logisim result looks better.
But the work you have to do to make both looking good again, is nearly the same.
   But also: I think that's a question of habit.

- _Would there be a way in Digital to place the components with the simulation active (initial state only), but automatically reset the simulation when pressing "start"?_
	This contradicts the architecture of Digital: In Digital a simulation model is generated from the circuit, every time the circuit is "switched on". It is not intended to modify this model while it is running.
    But i will think about it. Can you explain in more detail which use case you want to improve?

guiweber, 2017-05-26:
- _Moving wires and components_
I now better understand the design choice that was made regarding wires. As for the CTRL+Click, while it is documented, in my opinion it does not follow the _Principle of least astonishment_. However, considering that wires lose their connections when moving single wires, I agree with you that it is rarely useful to move a single wire and it doesn't really matter how easy or intuitive it is to move a single wire.

But there may be a balance to strike between the current system and implementing full wire routing. What about making straight wires (that were placed in a vertical or horizontal orientation by the user) remain straight and keep their connections when moving a component with wires or when moving a single wire? This would go a long way in making component movement easier and making the circuit look nice after a component has been moved. It would certainly reduce the number of wires that have to be drawn again after moving a component. Angled wires could keep the same behavior and change angle when moved in relation to a component.

- _Simulation during component placement_
There are several use cases. For students, it allows them to see immediately if there is an outstanding issue with the circuit when they add a component (e.g. a short circuit).

Also, it can be useful if you have a large circuit and want to modify/debug something that affects only a single state of the circuit. Logisim allows you to set your circuit in the right state then to make changes and see how it affects the circuit without restarting the simulation and having to re-set your inputs every time.

hneemann, 2017-05-26:
_Moving wires and components_
Maybe it is worth to rethink the way wires are drawn and components are moved.
Today i experimented with drawing lines always horizontal and vertical and the possibility to switch to diagonal mode if needed. This looks good so far.

_Simulation during component placement_
A input can have a default state which is set, when the model is started.
And there is a function "Set Inputs" in the Edit menu which reads the actual state of all inputs in the running model and makes them the new default state. This may help in the latter case.

hneemann, 2017-05-27:
Hi @guiweber 
I have added a undo function. It was simpler than i thought.
If you use Digital it would be nice if you could use the HEAD revision to play around with the new functions. If you are not able to build digital from the sources you can use [this](https://drive.google.com/open?id=0B1nBRhIo-nQuWmVyZlo5cFV0QzA) version.

guiweber, 2017-05-27:
Awesome : ) Thanks @hneemann !