hneemann/Digital #10: Checkbox visible in Background Color section of the Edit Circuit Attributes window

guiweber, 2017-05-26:
In the Edit Circuit Attributes window, one of the two checkboxes (DIL package, Modification Locked) appears in the Background Color section. The checkbox appears only after moving the mouse over the Background Color section. The checkbox that appears is the last one over which the mouse moved over.

[image]

hneemann, 2017-05-26:
Not reproducible on my machines. But i think it is a problem with the use of transparent colors.
Now I fill the button first with non transparent white and afterwards the with desired color.
Hope that helps. v0.12

closed 2017-05-26 by hneemann

guiweber, 2017-05-27:
@hneemann Since this was not reproducible on your machine, I just want ton confirm that the issue is fixed in the new build you provided. Cheers!