hneemann/Digital #12: Issues when importing a circuit with unnamed pins

guiweber, 2017-05-26:
1. The error message says the same thing twice:
[image]

2. A large text object is put on the schematic. This is unnecessary as there is already an error message pop-up giving the same information to the user.
[image]

hneemann, 2017-05-26:
You are right! The text is quite large. 
But something must be drawn, because the user must be able to select and remove it, if needed.
Any suggestions?

guiweber, 2017-05-26:
Why do you need to draw something? Can't you just cancel the component placement after displaying the error pop-up message?

hneemann, 2017-05-26:
This situation can also happen if you modify files on disk. For example, opening a file which embeds other circuits which contain now errors that were not present when the file was created. You should be able to do something in such a case. Simply removing all erroneous circuits feels bad.

guiweber, 2017-05-26:
Ok, then maybe skipping the pop-up in this case would be fine since the error message will be displayed in the schematic.

I feel a shorter message such as "Error importing test.dig: unnamed pin" would suffice. It would also look nicer if the selection oval didn't obstruct the message. Also note that the current error message displays the full path, which can make the message quite long. I'm not sure if this is really needed, maybe just the file name would suffice.

hneemann, 2017-05-27:
used a much shorter text in the "missing shape" shape.
added in v0.12

commit c43cc94c (2016-12-03): fixed #12
simplified test data with new test data parser
closed 2017-05-27 by hneemann