hneemann/Digital #14: Allow skipping pins in the Pin Wizard

guiweber, 2017-05-26:
It would be useful to be able to skip pins in the Pin Wizard for unused pins such as GND and Vcc. 

Suggestion: Right clicking could skip to the next pin number.

hneemann, 2017-05-26:
Click twice to skip a number.

closed 2017-05-26 by hneemann

guiweber, 2017-05-26:
Thanks. Is there a place where the controls/shortcuts are documented?

guiweber, 2017-05-26:
Nevermind, I found the help document in the installation folder. I guess the double click will be added at a later time, although it would be nice if the shortcut were displayed in the pin wizard itself.

Also, it might be a good idea to allow opening the documentation from the app itself. It could be placed under Help->Documentation.

hneemann, 2017-05-26:
There is no short cut at all: Clicking on a pin assigns the number to that pin and then increases the number. There is no check if a pin has already a number or not. So you can assign all numbers to the same pin if you want. The last assigned number wins.

guiweber, 2017-05-26:
Ah, I see. Yes, it is logical. I thought you meant double-clicking (quickly) would increase the number. It might be good to document how the pin wizard is intended to be used for people who are a bit slow like me ;- )