hneemann/Digital #18: Suggestion: Allow inverting gate inputs without using inverters

guiweber, 2017-05-27:
This is something that I love in Logisim but can't do in Digital. While it is purely a convenience feature, it helps make circuits a lot more compact. It is also less wires to connect.

[image]

Logisim implements this with a drop-down menu for each input (inverted yes/no), but if you want to keep the UI simple and compact a simple text field where a comma/space separated list of input to invert can be provided would work very well.

p.s. I will stop sending you lots of issues soon ;- ) I worked all day in Digital yesterday hence all the issues, but now I think I have pretty much found all I could. Thanks for your work and your support!

hneemann, 2017-05-27:
I thought about this in the past, and up to now i have no idea how to implement that in an elegant way.

p.s: Never mind. I aways appreciate suggestions and opinions from other people. This helps me to improve my software. 
And by the way: It would be nice if you could tell me at which school/college/university you plan to use Digital. If you prefer a private message use [digital-simulator@web.de](mailto:digital-simulator@web.de). And of course it's okay if you do not want to tell.

guiweber, 2017-05-27:
Do you mean you didn't find an elegant implementation codewise or in relation to the UI?

hneemann, 2017-05-27:
This affects the code which creates the simulation model based on the schematic, which is much more complicated than the minor GUI changes that you have suggested in the recent days.

hneemann, 2017-05-28:
Playing around with the code it seems to me that it is very hard to implement the input inverter in a general way, but its maybe possible to add this function to the basic gates and the flip-flops, which also solves #16.
It will never ever be implemented for the usage on nested sub circuits, and also the analysis of sequential circuits with an inverted clock input at one of the flip-flops will never be implemented.
In the next days i will try to implement a prototype.

guiweber, 2017-05-28:
Sounds good, thanks for looking into it. I think what you propose would be sufficient to make most circuit designs a lot more compact and look more similar to the circuits found in datasheets and such.

hneemann, 2017-05-31:
added in v0.12

closed 2017-05-31 by hneemann