hneemann/Digital #19: Visibility of circuit errors

guiweber, 2017-05-27:
When a circuit error happens (such as no value set for an input), a message is displayed and a circle is displayed around the affected component. However, if the component is outside the view of the schematic, then the user will not be able to see which component was affected. The x, y coordinates are provided in the error message but these are not useful for the user since there doesn't seem to be a way to display them in the schematic.

I suggest centering the schematic view on the affected component when such an error happens (but make sure the pop-up is not over the component). I would also remove the x, y coordinates from the error message since they are not really useful.

Also, making the error circle around the component bright red would make it more obvious and different from the regular selection circle.

hneemann, 2017-05-27:
This could be confusing because if you switch to a new location in the circuit the user can no follow.
Maybe its better to zoom out to the maximum extend so you can see the error location.
Or only zoom out if the error is not visible?

guiweber, 2017-05-27:
These could be good solutions too!

Another I just thought of could be to make the error circle persist after the pop-up has been dismissed. That would allow the user to easily find the component.

engstad, 2017-05-27:
I enjoy this program as I am teaching my daughters some basic digital electronics. 

That said, they have a problem differentiating wires and buses. It would be great to use a different scheme for this, perhaps indicating a bus with a different style? Such as `=` versus `-`?

They also had some problems seeing where to hook things to. Perhaps you could lengthen the "pin" part of the components?

hneemann, 2017-05-27:
@guiweber: Thats a good idea, but when should the circles be removed?

@engstad: This is a good hint. Also my students sometimes have problems to recognize that. So how to make clear that a line is not a simple wire, but a bus?

guiweber, 2017-05-27:
@hneemann It could be removed when the user interacts with the component in any way or starts the simulation again.

hneemann, 2017-05-27:
Added in v0.12.

closed 2017-05-27 by hneemann