hneemann/Digital #20: Additional shortcuts and time saving functionality

guiweber, 2017-05-27:
After working some more in the latest version, I would suggest the following:

Shortcuts when a basic gate is selected:
- PLUS (+) : Increase the number of inputs by one
- MINUS (-) : Decrease the number of input by one

For wire placement (which, by the way, is much improved in the latest version!):
- When pressing CTRL+Click to select a wire while in wire placement mode, get out of wire placement mode and select the wire instead. It happened a few time that I was trying to select a wire and it wouldn't work because the wire placement mode was active. When it happens, it can hard to see since the wire being placed is over the wire you're trying to select. CTRL+Click should have the priority since it is a more elaborate command which can hardly be made by mistake.

For components that have a label: 
- When you open the component properties, it would be nice to give focus to the Label field, so that you can just type right away and press enter to confirm the new label name. At the moment, the Rotation field has the focus by default. However, rotation can be done with the R shortcut directly in the schematic, so you will rarely want to open the properties dialog to change the rotation. Giving the Label field focus by default would make more sense.

hneemann, 2017-05-27:
added in v0.12

closed 2017-05-27 by hneemann

hneemann, 2017-05-28:
@guiweber 
_For wire placement_
I decided to draw "wires in progress" in the highlighted color. That makes it easier to see that there is a "wire in progress" at all.