hneemann/Digital #21: Improved graphical distinction of wires and buses.

hneemann, 2017-05-27:
One should clearly see if a connection is a simple wire or a bus. 
Color discrimination is not sufficient. Maybe one could draw several parallel lines?

hneemann, 2017-05-28:
Very hard to implement! 
In draw mode there is no information about the number of bits on the wire. Only if a model is created and the model is running, the number of bits is available. And if the model is running there is already a special color indicating a bus (blue). Also a wire can not have a state, because wires are something very fluent. Wire objects are merged and replaced by other wire objects all the time. So its hard to keep the bit number from the past model run. Also this approach will led to an inconsistent drawing sooner or later.
The only thing I can do is to change the visual representation in run mode. 
Does this make sense? It does not help while drawing the schematic.

hneemann, 2017-05-29:
@engstad: I played a bit with the buses:
Drawing two parallel lines as suggested looks awful: Two thick lines make the buses looking extremly wide which does not work well with the pins, drawing two thiner lines makes them appear less important. Also increasing the width does not work well with the pins.
As long as no one has a really good idea, I'll leave it as it is.

closed 2017-05-29 by hneemann

engstad, 2017-05-29:
Interesting. I suppose you would have to also add thick bus input and output lines for it to look good.

Is another possibility to do "cross-marks" like you see sometimes on schematics?

Finally, you could go the Kicad route, by marking them in dark blue.

commit f64971f2 (2017-10-27): added bit info to wires, fixes #21
commit 7decf0db (2017-10-28): added bit info to wires, fixes #21