hneemann/Digital #22: Add Transmission Gates

engstad, 2017-05-28:
For implementing XOR, you can use 6 transistors, two of which are transmission gates. It would be great if you could add those!

hneemann, 2017-05-28:
There are already transmission gates. 
They will be included in v0.12.
If you want to try them now, you can build head from the sources or you use  [this](https://drive.google.com/open?id=0B1nBRhIo-nQuWmVyZlo5cFV0QzA) pre release version.

closed 2017-05-28 by hneemann

engstad, 2017-05-28:
Awesome!

hneemann, 2017-05-28:
And by the way: In Digital the transistors are simulated without a body diode.
This means in Digital you can build a transmission gate with only one transistor. ;-)

hneemann, 2017-05-29:
@engstad: Do you mean the following circuit?
[image]
The simulation of this circuit does not work in Digital.
And this is the reason: Imagine the state as shown above. When A goes high, A is connected directly to B via B's PFET and the transmission gate, and B is in low state. Although this is allowed during step calculation and does not cause an error, the resulting state is not defined and it may be, that B's low wins, and also the A line stays in low state.
In this case the xor is not working. There are some other situations which cause similar problems. So the behavior of this circuit is not predictable in Digital. Sorry for that.
The root cause of problems like this is that Digital does not simulate the resistance of the closed switching components such as FETs. These components always have zero resistance if closed. If you are trying to build a circuit that relies on the resistance, it will not work. Maybe you should try LTspice instead.

engstad, 2017-05-29:
Yes, that's the one. And I tried it too, hoping it would work, but it didn't, as you mentioned.

Ah well, it's not that important - thanks for looking into it! I still really enjoy your program!

By the way, have you looked at Everycircuit? Their GUI is quite remarkable.

hneemann, 2017-05-29:
@engstad: 

_Does this also have anything to do with "weak" 0s and 1s?_

Thats a really tricky question. Its more a limitation of the simulation engine. You have to make a decision: How many details of the circuit should it cover, keeping in mind that, the more details it simulates the slower the simulation becomes.
The main goal of implementing a new simulator was, to make it fast enough, to run an assembly program in a simulated processor with sufficient performance to use that processor in my lectures on systems programming.
So I decided to leave anything out, which is not necessary to simulate a processor. Therefore, there are several limitations:

- A clear distinction between inputs and outputs, with no feedback from an output to an input or from on output to an other output. This makes a simple resistor impossible. A switch is an exception: In case of a switch the wires are simply shorted out, if the switch is closed.
- Every gate has the same delay time, which also improves the performance.
- The circuit is analyzed and a really simple "flat" simulation model is created. This also improves the performance, but does not allow to edit the running model. Also the startup takes some time.
- In Digital a weak 0 or 1 is a net state: If all outputs connected to a net are in high z state, the nets output is set to 0 or 1 instead of also became high z. If you connect in pull up resistor to a net this simply sets this default value to 1. In a real circuit this is more a property of an output itself (open collector).

As a result of the decisions made, the simulator can run a simple processor at 100kHz which is sufficient for my needs, but it can not simulate a 6 transistor xor.

hneemann, 2017-06-23:
@engstad 
With a little hack I could bring your xor to work. :-)
Try [this](https://drive.google.com/open?id=0B1nBRhIo-nQuSHNRcVMyZkpTQWc) pre release version and this circuit: [xor.zip](https://github.com/hneemann/Digital/files/1098850/xor.zip)