hneemann/Digital #23: High DPI scaling not working under Linux

guiweber, 2017-06-06:
I just tried running Digital under Fedora 25 with Gnome 3.22.3 and Wayland and unfortunately the DPI scaling isn't working.

hneemann, 2017-06-06:
In the moment I get the screen resolution by the appropriate Java API.
It seems to me that sometimes a wrong value is reported.
As far as i understand you can correct the value reported using the Java ControlPanel. Please try that first. If this does not work, you can try this [pre release version](https://drive.google.com/open?id=0B1nBRhIo-nQuMHJQSUtXWkZ2aTA) which also allows to override the reported value by a new setting. The default value you find here is the resolution reported by the Java API.

hneemann, 2017-06-06:
I digged down the problem and found out that to get the screen resolution, openJRE simpliy calls `DisplayWidth()` (returns width in pixels) and `DisplayWithMM()` (returns width in mm) from the Xlib and then gets the screen resolution by calculating `DisplayWidth()*25.4/DisplayWithMM()` which gives the resolution in DPI.
To get a wrong resolution this way, there must be a problem in Xlib or below that. I have no idea what happens when these functions are called on a Wayland system, where XWayland is used to run a X-Server based on Wayland to allow applications still using Xlib.

My conclusion: Set the resolution manually in the settings!

guiweber, 2017-06-06:
Thanks. Manually setting the DPI works well.

That being said, automatic DPI scaling has worked well for a while on Fedora with Gnome, so I suspect there is a reliable way to get the screensize and resolution. I will test under a native X session to see if the issue is really related to Wayland. Where do you store the settings for Digital on Linux so that I can delete them and start anew?

hneemann, 2017-06-06:
The settings are stored in your home folder in a file named ".digital.cfg".

guiweber, 2017-06-06:
Thanks. I just tested under X11 and it also starts at the default value of 96DPI. It looks like the issue is not specific to Wayland.

hneemann, 2017-06-06:
At May 11th, you told me "The font size and icons are now the perfect size." See #6.
I checked the commit logs: I have not changed anything on the DPI calculation since.
So what have you changed since it was working?

guiweber, 2017-06-06:
Oh, sorry for the confusion! When I filed #6, I was on Windows 10. When I said that the font size and icons are the perfect size, it was still on Windows 10. I didn't check on Linux to see if it worked at the time.

Today I needed to work under Linux and it is only then that I noticed that the DPI scaling doesn't work. 

So there is no regression, it probably just never worked on Linux. It still works fine on Windows 10 : )

hneemann, 2017-06-07:
So I will keep the new dpi setting in the code.
But for now, i will show it only on Linux.

closed 2017-06-07 by hneemann

guiweber, 2017-06-07:
It might be good to show it on Mac as well, since it hasn't been tested and the automatic scaling may not work for all we know.

hneemann, 2017-06-08:
Could you please test [this](https://drive.google.com/open?id=0B1nBRhIo-nQucTZiOUs3aENvMHc) version? I have changed the default font scaling and I'm not sure if it still works on Win+Hires.
Most users will understand "font size" better than "screen dpi".

guiweber, 2017-06-08:
It still works on Windows 10 and still doesn't work (by default) under Linux.