hneemann/Digital #24: Resize der Baumansicht der Bauteile klappt nicht immer

carstenhag, 2017-06-08:
Version: `v0.12.1`, commit `93a576c2e193caa54b9e29ae8341af42919f9cca`
OS: macOS Sierra, 10.12.4 - MacBook Pro, Retina 15'

Die Baumansicht der Bauteile wird angezeigt wenn man unter Ansicht diese Option aktiviert. Sie öffnet sich aber zu klein und man kann die Größe nicht immer verändern.

Die Veränderung dieser "Vertical Bar" funktioniert nur bei jedem ca. fünften Versuch (beim verschieben nach rechts). Wenn man die Bar nach links verschiebt, klappt es immer.

Screenshot:
[image]

hneemann, 2017-06-08:
Können Sie Digital einmal auf der Komandozeie mit folgender Anweisung starten:

`java -Dswing.defaultlaf=javax.swing.plaf.metal.MetalLookAndFeel -jar Digital.jar`

Wie sieht es dann aus?

carstenhag, 2017-06-08:
Scheint sich nur Optisch geändert zu haben, Bug bleibt.
[image]

hneemann, 2017-06-08:
Die Icons in der Baumansicht scheinen aber jetzt unterscheidbar zu sein?

carstenhag, 2017-06-08:
Ah ja, das habe ich total vergessen. Die Icons sieht man jetzt gut.

hneemann, 2017-06-08:
Können Sie einmal [diesen](https://drive.google.com/open?id=0B1nBRhIo-nQucTZiOUs3aENvMHc) Zwischenstand starten und schauen, was im Settings Dialog bei "Schriftgröße im Menü" angezeigt wird?

carstenhag, 2017-06-08:
Momentan steht dort 114%.

hneemann, 2017-06-08:
Und der Rest (Icons, Schriftgröße) ist so weit ok?
Sind die Icons in der Baumansicht auch noch sichtbar, wenn Sie Digital per Doppelklick starten (also ohne das `java -Dswing.defaultlaf=javax.swing.plaf.metal.MetalLookAndFeel -jar Digital.jar`)?

carstenhag, 2017-06-08:
Icons und Schriftgröße sehen gut aus.

Mit beiden Startmethoden (mit dem -Dswing... flag und ohne) sind die Icons sichtbar.

Bei der Startmethode ohne dem Flag wird mit dieser neuen Version allerdings der Haken bei dem Menüeintrag angezeigt, auch wenn die Ansicht deaktiviert ist:

[image]

hneemann, 2017-06-08:
Wird der Haken konsistent, wenn Sie ihn klicken? Passen also Haken und "Vorhandensein der Baumansicht" nach dem ersten Klick zusammen?

carstenhag, 2017-06-08:
Nein, dieser bleibt immer "aktiv". Hatte ich vergessen zu schreiben.

hneemann, 2017-06-08:
Wenn Sie noch Lust haben, können Sie einmal die Schriftgröße auf 100% setzen und nach dem Neustart noch einmal das Häkchen ausprobieren.

carstenhag, 2017-06-09:
Digital wir mit `java -jar Digital.jar` gestartet. Neue version (die hier verlinkte).

Wenn die Schriftgröße auf 100% gestellt ist, hat Digital das gleiche "Skin" wie mit `-Dswing.defaultlaf=javax.swing.plaf.metal.MetalLookAndFeel`, komischerweise. Dort ist die Checkbox/der Haken korrekt.

Mit 114% wird das "normale" Skin angezeigt und es tritt der Bug mit dem Haken auf.

hneemann, 2017-06-09:
Danke für die Info! Können Sie mit der neuen Version Einfluss auf die Schriftgröße nehmen? Ändert sich überhaupt etwas an der Schriftgröße, wenn Sie z.B. 200% einstellen?

carstenhag, 2017-06-09:
Ja, die UI wird riesig.

[image]

carstenhag, 2017-06-09:
Mit der neuen Version und mit 114% Schriftgröße funktioniert der Haken.

[image]

Der originale Bug des verschiebens der Vertical Bar existiert noch.

hneemann, 2017-06-09:
Da bei der JSplitPane die Java-Swing-API ohne irgend welche Anpassungen verwendet wird, vermute ich, dass das Problem nicht in Digital zu suchen ist ([Siehe hier](https://www.heise.de/mac-and-i/meldung/Java-Aerger-mit-macOS-Sierra-3567822.html)). Daher werde ich hier zunächst nichts weiter unternehmen. 
Vielen Dank für Ihre Hilfe.

closed 2017-06-09 by hneemann