hneemann/Digital #27: Bidirectional pins.

Akari1982, 2017-06-19:
Is it possible to create bidirectional pins that can be used an IN pint and as OUT pins simultaneously? This is usual for data buses.

hneemann, 2017-06-19:
By default every pin is bidirectional.
There are inputs and outputs, but this almost only effects the shape of the pin.
If you include a circuit as a subcircuit all pins are bidirectional.

Akari1982, 2017-06-19:
Hmm. This is working in top cirquit, but if i test it in subsircuit I still get an error because input pin emits some value. Maybe posibility to set it to "not connected" state help to fix this.

hneemann, 2017-06-19:
Enable tristate values at the input pin. Then you can set it to the tristate value, which means "not connected".

Akari1982, 2017-06-19:
Is it possible to set pin in "not connected" state by default?

hneemann, 2017-06-19:
This is not possible. 
But I will add that feature.

closed 2017-06-19 by hneemann

hneemann, 2017-06-20:
@Akari1982 
I added this feature. You can checkout HEAD and try it if you like.

Akari1982, 2017-06-20:
Thanks! You are very fast =)

hneemann, 2017-06-20:
Only at issues that are very easy to implement. ;-)