hneemann/Digital #29: FET in some configuration doesn't work.

Akari1982, 2017-06-20:
I tried create CMOS full adder cell from PSX CPU and encount error. When I try switch pin B or C in given circuit from 0 to 1 not order of update of elements rise error.

At picture there are full adder on left and simplest part that generated error on the right. 
[image]

If i set default state from the start - circuit updates correctly, but it fails again if I switch from 0 to 1.

hneemann, 2017-06-20:
Haven't looked into it in detail but it seems a duplicate of issue #22.
In Digital you can not build a circuit that relies on a residual resistance of a conducting FET. The resistance of a conducting FET is always 0 Ohms. So here can not be a different voltage on Source and Drain if the FET is conducting.

Akari1982, 2017-06-20:
By the way same stuff is working on Logisim.
[image]

Akari1982, 2017-06-20:
It's more unusual stuff in half adder. Sometimes it's working, but sometimes I got same error as in full adder. Can't get dependancy. Error rise when I switch from 1 to 0 sometimes. But next time it can be all right.

[image]

hneemann, 2017-06-20:
Unpredictable behavior is what I would expect: If a conducting FET shorts out two inputs which have a different state (one input is high, the other is low), the result is not predictable: The line is either high or low.

hneemann, 2017-06-20:
And by the way: As far as i understand in Logisim a transistor is not a switch. Its a logic gate with two inputs (gate, source) and one output (drain). If you look at a N-channel FET, the function is mostly:
```
if (gate==HIGH) 
   drain=source
else
   drain=HIGHZ
```
So it is not possible to build a SRAM cell, because this requires a "bidirectional" FET which allows the flow of current from source to drain and vice versa.
So Logisim and Digital both have their limits, but they are not the same. ;-)

If you want to build a circuit in Digital that behaves like a circuit in Logisim, you can use Digitals driver component and the driver with inverted selection instead of the FETs. As far as I understand, the drivers in digital have the same behavior as the transistors in Logisim. But I have not tried it.

hneemann, 2017-06-20:
Not possible to fix because there are no real resistors.

closed 2017-06-20 by hneemann

hneemann, 2017-06-21:
@Akari1982
Maybe you can use this more conventional full adder:
[image]
[cmosFullAdder.dig.zip](https://github.com/hneemann/Digital/files/1090624/cmosFullAdder.dig.zip)

Akari1982, 2017-06-22:
Thanks. I know that i can create adder without errors, but i'm not creating it from scratch. I just reproduce real cell from PSX cpu. http://wiki.psxdev.ru/index.php/CPU_MAC
I think FETs are supposed to work like real one, because otherwise i can just use ANDs and ORs to create same thing.

hneemann, 2017-06-22:
The simulation technique used in Logisim and Digital - which are very similar by the way - are not able to simulate a FET. This is because both simulators only take into account the local environment of the components to calculate the outputs. This means: Take the input state of a component and then calculate the output of that component.
But to simulate a real FET you have to take into account the whole circuit. You have to transform the whole circuit in a set of equations and you have to solve the equations to get the output state. This is how [LTspice](http://www.linear.com/designtools/software/) works. The disadvantage of this approach is that it is very slow. You can not run a processor in such a simulator.
So Digital and Logisim can only simulate something that in most cases behaves like a real FET. And both simulators have different edge cases which don't work as expected. Both are designed to simulate digital circuits. The capabilities to simulate almost analog circuits are very limited.

Akari1982, 2017-06-22:
Yeah, i understand simulation logic. And i think simplest case like this can be solved by update order. Right now FET pass value earlier than update it's gate. This cause values conflict. If you update passing of values through FET at tha same time as update their gate - this exact problem will be solved, because "valve" will be closed.

hneemann, 2017-06-22:
I don't think so. In Digital if a FET is conducting, source and drain are shorted out. So source and drain can not have a different value anymore. And this means, that is some cases the common line does not change, even if one side of the FET changes its value. It just does not happen anything at all. And this is not dependent on the order of calculation or the timing.
But its open source: Feel free to play around, implement something and try out! :-)
There are a lot of CMOS test cases: You will notice immediately if something breaks. :-)

But if you are interested in CMOS circuits its maybe a good idea to use a spice based simulator instead.

hneemann, 2017-06-23:
@Akari1982 
For performance reasons, there have long been unidirectional switches. These were always used when one of the switch inputs is a constant. I've added an option to the FET that forces the use of these unidirectional switches, even if none of the inputs is a constant. If this new option is set, the mirror full adder, the GDI full adder, the SERF adder and the 10T full adder are working.
This option does not solve the problem, but its maybe a work around, sufficient for most problems.
Please let me know if your circuit works with this option set. You can find a pre release version [here](https://drive.google.com/open?id=0B1nBRhIo-nQubHk2SVlFRzNtYnM).

Akari1982, 2017-06-23:
Yup. This helps. At least this is wirking like Logisim =)
https://gyazo.com/94f5d36f09b8909d1df808d9c1f6838e

hneemann, 2017-06-24:
The source is on the same side as the gate.