hneemann/Digital #31: No T input for T flip-flops

guiweber, 2017-06-22:
T flip flops should have a T input that enables the toggle function of the flip flop.
https://en.wikipedia.org/wiki/Flip-flop_%28electronics%29#T_flip-flop

While some real world T flip flops may omit the T input, having the input in Digital would be better since it can be hooked to a constant when not needed and would allow simulating T flip flops that do have the T input without using other logic components.

hneemann, 2017-06-22:
Why not using a JK flip-flop with J and K shorted out?

[image]

guiweber, 2017-06-22:
In my case, this would be more confusing for students since they learn about the T flip-flop first and have to use it before learning about the J-K flip-flop.

hneemann, 2017-06-22:
Thats a good point. I will provide a T-FF with two inputs is some way.

hneemann, 2017-06-22:
added in v0.13

closed 2017-06-22 by hneemann

guiweber, 2017-07-16:
@hneemann Hi, just wondering when you plan on releasing V0.13 to take advantage of the new features.

hneemann, 2017-07-16:
@guiweber I need no new release until October, but i can create a new release if needed.

I have a question concerning the 74xx circuits: At the the moment there are no pins for VCC and GND. Would you rather omit them, or do you think it is easier for the students to have these inputs, and to enforce that they are connected properly?

guiweber, 2017-08-30:
@hneemann would it be possible to make a new official release this week?  Classes start next week and I would like the students to have access to the latest changes.

hneemann, 2017-08-30:
@guiweber: I have planned a new release this week.
But the T-FF is already included in v0.13, which has been available for about a month.

guiweber, 2017-08-30:
@hneemann Yes, I have been using 0.13 as well as the Beta versions myself. I was only replying on this thread because that's where we talked about new releases ;- )

Thanks!

hneemann, 2017-08-30:
@guiweber Could you provide me with a list of 74xx circuits you plan to use this term?
I'm not very confident about all of them. Some of the more complex circuits are lacking test cases.

guiweber, 2017-09-01:
The ICs the students are going to get are below. They might also get other random parts in addition to these, so the more we have in Digital the better.

7400
7402
7404
7408
7410
7420
7432
7474
7480
74109
74112
74138
74151
74163
74164
74191
74238