hneemann/Digital #32: Import error appears even after removing the error causing part from circuit

guiweber, 2017-07-16:
Steps to reproduce:

- Save a circuit named test.dig in that contains only basic parts under \foo and close digital
- Put the 74116.dig file in \foo\a and in \foo\b
- Open test.dig and try adding a 74116 IC to the schematic. This will trigger an error because the same file exists twice in the import path.
- Dismiss the error pop-up and delete the part (large error message placed in the schematic) so that no trace of the 74116 part exists.
- Try saving the document again and notice that the error pop-up appears again.

Expected results: Removing the error causing part from the schematic should prevent the error message from appearing.

guiweber, 2017-07-16:
Another issue that seem to be linked to this:

- Save a circuit named test.dig in that contains only basic parts under \foo and close digital
- Put the 74116.dig file in \foo\a and in \foo\b
- Open test.dig and select File --> Save As  and save test.dig to \foo\b
- Try adding a 74116 to the schematic and notice that it still produces an import error

Expected results: Saving the file at a new location should refresh the files in the import path

hneemann, 2017-07-16:
The first issue is easy to fix but the second one is not as easy.
If you save a file with "save as", the library path is only set if none has already been set. This is necessary because there is nothing like a project or a project file: If a file is opened the location of this file defines a project and the library path which is used. 
To change the library path at "save as" is very confusing if you are rearranging a circuit consisting of different subcircuits.
I have to think about it.

hneemann, 2017-07-16:
fixed in V0.13

closed 2017-07-16 by hneemann