hneemann/Digital #34: Error when running tests

guiweber, 2017-07-16:
I'm trying to add test cases for a new IC. I'm not sure what I'm doing wrong, but I get the following error message. The input pin labels match the names in the test cases.

[image]
[image]

hneemann, 2017-07-16:
Tabs are not allowed in the header line. I will fix that.

hneemann, 2017-07-16:
fixed in v0.13

closed 2017-07-16 by hneemann

guiweber, 2017-07-16:
Ah, I see. Thanks.

Also, how would you go about testing outputs that should keep their previous values or previous values of other input or output?

For example, the 74112 has three such states: 2 "no change" states where the outputs keep their previous values and one toggle state where the output takes the value opposite to its previous value. I tried writing the test cases like this but it doesn't work:

~1CLR ~1PR 1J 1K ~1CK 1Q ~1Q
1 1 0 0 C 1Q ~1Q
1 1 1 1 C ~1Q 1Q
1 1 X X C 1Q ~1Q

hneemann, 2017-07-17:
"Keep value" and "flip value" is not available during testing.
You can set a defined state, and than test the transition:

~1CLR ~1PR 1J 1K ~1CK 1Q ~1Q
1 1 0 1 C 0 1
1 1 0 0 C 0 1
1 1 1 1 C 1 0
1 1 0 0 C 1 0
1 1 1 1 C 0 1

If this is not sufficient you can use loops do define test cases (see cmos/sram.dig).

guiweber, 2017-07-17:
Ah I see, so each new test case starts with its output initialized with the values set in the previous test case.  I was under the impression that the test cases were independent from each others. It might be useful to expand the documentation to explain they are not and how sequential behavior can be tested. Cheers!

hneemann, 2017-07-17:
@guiweber At every line at first the input values are set, then a clock cycle is performed if a "C" is present in the line and after that the output values are checked.
But the model is initialized only once before the first line is applied, so it preserves its state. If you are testing a pure combinatorial circuit without a state, every line is independent from the others.

hneemann, 2017-07-17:
@guiweber And by the way: I have had good experiences by providing a circuit with only some test cases and the necessary inputs and outputs to the students and let them implement the required circuit. 
So i provide an exact specification of what is needed and during the exercise the students can always check if the circuit works like expected or not.