hneemann/Digital #36: Mandatory connection of all input pins for DIL package

guiweber, 2017-07-17:
In order to run the simulation, Digital requires all inputs on DIL packages to have a connection to a definite state. I understand how it is necessary to have all inputs connected for the simulation to be able to establish the state of the outputs. However in practice, when using ICs in DIL packages, sometimes some outputs and their associated inputs remain unconnected because they are not needed. For example, you might only need to use one of the two flip-flops on a 74112.

This makes it impossible to make circuits in Digital that look like what you would really do. For example, in practice, a circuit using a 74107 to toggle an output would look like this. The inputs and outputs of the second flip-flop are not connected but it works just fine.
[image]

While in Digital you need to connect all other pins to an arbitrary value, which makes it look like this:

[image]

Would there be a way to remove the requirement to have all inputs connected for DIL packages? In order to avoid dealing with undefined states, this could be effective only if the associated outputs are also not connected. I haven't seen the code so it might not be that simple, but I'm thinking that if all outputs of a circuit inside a DIL are not connected, you can simply ignore that circuit during the simulation.

I just feel the additional wires that need to be connected are causing unnecessary clutter in the schematics, take additional time to setup and might be confusing for the students.

hneemann, 2017-07-18:
Floating inputs are something you should avoid. Even if you build a real circuit. It is a bad practice not to connect unused inputs.

In CMOS ICs this can lead to oscillations or to an unexpected high current consumption in the input stage. Under certain circumstances the high current can damage the IC. 

Only TTL-ICs which are using a multi emitter transistor in the input stage are able to deal with open inputs which act like connected to high. And also in this case its recommended to connect unused inputs to high to avoid noise.

See [here](http://www.ti.com/lit/an/scba004d/scba004d.pdf), [here](https://www.classe.cornell.edu/~ib38/teaching/p360/lectures/wk09/l26/EE2301Exp3F10.pdf) and [here](http://www.ti.com/lit/an/sdya009c/sdya009c.pdf).

I do not want to implement anything, just to enable something that should be avoided in practice.

closed 2017-07-18 by hneemann

guiweber, 2017-07-18:
Thanks for the references, I hadn't thought about all aspects of the issue. I agree with you then.

hneemann, 2017-07-18:
@guiweber What do you think about a mandatory connection of the power supply? (See #31). Would this be a help for your students to build a real circuit? You can try the actual [pre-release](https://infdigital.dhbw-mosbach.de/): There is a function in the 74xx menu to add a power supply.

guiweber, 2017-07-19:
I don't really mind if the VCC and GND pins are connected or not, I think that powering the ICs is something that the students understand easily. VCC and GND connections might be nice if we could use them to switch parts of the circuits on/off, but in the current implementation it looks like the power supply must always be powered so there is, in my opinion, no real advantage to have it.

Also, since unused inputs all need to be connected, I was wondering if you have an idea of something that could be done to make it easier and faster to wire those inputs to VCC or GND. Maybe adding a shortcut would make it better? For exemple "ctrl-click" on a pin could connect it to VCC and "alt-click" could connect it to GND. Thoughts?

hneemann, 2017-07-20:
In my experience, most students using Digital, will not become a power user. ;-)
Most of them will never use a special short cut. Therefore, it is not necessary to implement such specialized functions.

Regarding the VCC and GND wires, I thought it would be easier for the students to build the real circuit if all the necessary wires were present in the circuit diagram, and not just part of it. But I have no experience with such exercises.

guiweber, 2017-07-20:
Well I will be making a video tutorial for Digital for the class, so I will show them the shortcuts : ) Otherwise most would definitely not use them, as you said!

I'm just concerned that moving from Logisim where you didn't have to connect all the inputs, it will take more time for the students to realize the same circuits in Digital because of all the additional connections. If you have a circuit with between 5 and 10 DIL packages where you  are using only 1 out of 2 or 1 out of 4 of the gates of the IC, that's a lot more clicks for the same results.

As for VCC and GND connections, it would maybe help them if they had to fully route the wires like they will be doing on the breadboard, as this would give them a good preview of the wiring. However in practice the VCC and GND pins in Digital will just be connected directly to the Supply Voltage and Ground components, and will not give a preview of the actual wiring. So I don't think they would be much useful, but as I said, I don't mind if they are there either.