hneemann/Digital #37: JAR Missing from latest release

shawty, 2017-07-26:
Hi just a quick note to let you know that the latest release has no JAR file in either of the archives, but your read me on the main repo page tells folks that it does and is ready to run without building.

All that exists in the Distribution folder is:
Digital.exe
ReleaseNotes.txt
Version.txt
Assembly.xml
l4jconfig.xml

When digital is run it fails and stops with a message it cannot find Digital.jar

hneemann, 2017-07-26:
Thanks for the hint. The release process is completely automated, just to avoid such kind of issues. 
But with the release of v0.13 something went wrong at travis.ci. :-(
I have restarted the travis.ci release process, and now everything seems to be there.

Digital is released as a [zip](https://github.com/hneemann/Digital/releases/download/v0.13/Digital.zip) file, containing all the examples and the binary. The whole zip file was missing. Sorry for that.

closed 2017-07-26 by hneemann

shawty, 2017-07-26:
No probs.  Happy to help, just thought I'd let you know as many like me may have come via read-me and wondered what's happening.