hneemann/Digital #38: Documentation: explain "nested circuits" better.

peterb12, 2017-08-18:
Although I have an intuition about what this probably means, it really isn't clear from the documentation.

hneemann, 2017-08-18:
I have updated the menu entry names and the tool tip texts. (See [here](https://infdigital.dhbw-mosbach.de/))
But the main problem is, that I am not an English native speaker! :-)
If you could provide a better translation/description, I would appreciate that. :-)

peterb12, 2017-08-19:
Happy to help translate.  Could you explain (in your native language is fine, I'll figure it out) what the intent is if the user selects "New nested circuit"?

My intuition is that what this means is (I'm not proposing you use this language, just confirming my understanding) "Create a new circuit (and new associated file) in this library", where "library" really means "In the same folder as the circuit I happen to be editing right now."

Is that correct?

hneemann, 2017-08-19:
I think so. But let me explain what is going on:

Digital uses a simple approach to deal with sub circuits: If you open a circuit you can import all circuits located in the same folder or in a sub folder via the "Components->Custom" menu. Only the file name matters, not the full path. Thus you can rearrange the folder structure without breaking anything. You also can copy all the files to any other location and it still works.

But there is a problem: This only works if the root folder is known. If there is no root folder, you can't import anything.

Imagine the following situation: You are working on a project and you decided to create a new sub circuit. You can do so by clicking "File->New". But then there is no file name yet. So you can't import a file from your project you are working on. This is a confusing situation. You have to save the new empty file first to define a root folder.

Therefore there is a menu entry "File->New nested Circuit" (I just renamed it to "File->New embedded Circuit"). In this case a new editor window is opened and the root folder remains unchanged. Thus you can import all the circuits you are working on.

hneemann, 2017-08-19:
Is in the main a duplicate of #33.

closed 2017-08-19 by hneemann