hneemann/Digital #41: Digital crashes if its folder name has some special characters

guiweber, 2017-08-18:
This was tested under Windows 10. Many (but not all) valid characters have been tested and only the following two are known to cause the issue:

Character that causes a Java Exception on startup: `! `
Character that prevents Digital from starting but does not cause an exception: `;`

hneemann, 2017-08-18:
I can not reproduce this bug, because I have no Win10 machine at hand.

Could you please start Digital with `java -jar Digital.jar` and send me the console out.

guiweber, 2017-08-18:
Here they are. Note that in contrast to the exe, opening the jar file works when the folder has a semicolon. However it still crashes when there is an exclamation mark.

[exclamation.txt](https://github.com/hneemann/Digital/files/1235352/exclamation.txt)
[Semicolon.txt](https://github.com/hneemann/Digital/files/1235353/Semicolon.txt)

hneemann, 2017-08-18:
It seems to me that this is an error in the XML libraries coming with the Java JRE.
So the only thing I can say is: Don't use semicolons or exclamation marks in folder names!

closed 2017-08-18 by hneemann

guiweber, 2017-08-18:
If possible, it would be good to detect when the path contains any of these problematic characters and display a more explanatory error message before the crash happens. As it is, it might be hard for someone to troubleshoot the issue.