hneemann/Digital #44: Issues with ROM component

roy77, 2017-08-22:
In attributes dialog address bit size takes effect after closing dialog. That is confusing when "Edit" is used before closing the dialog.
Before the "Edit" button opens the "Data" dialog a message box should allow the user to accept the current setting.

Can not work with 32 address bits.  Only one line of the memory is shown in "Data" dialog.

With data bits 25 and address bits 25 "Edit" do not open the "Data" dialog.

The data dialog shows only 6 chars per cell. So working with data bit sizes greater 24 is a pain. Each time the "Data" dialog is opened the column size of the interested cells have to be changed manually.

hneemann, 2017-08-22:
There a are some issues with the ROM data editor if the bit size is large. I will fix them. 

But if you really want to use such a large ROM with 32 address bits there will arise other issues: 
The XML file which is used to store the circuit will become extremely large if you store 4*10^9 values in such a ROM: The values are stored as hex values, the XML file can easily become larger than 20GBytes. Opening and storing such a circuit will become extremely slow. It will be unusable.

roy77, 2017-08-22:
I will only use a small part of the possible address range. I converted a simple 32-Bit RISC processor from logisim (16 bit address bits limitation) to  digital and omit the bit size reduction.

It´s OK for me if you restrict it to 16 or 24 bit.  I only use it because it was selectable.

hneemann, 2017-08-22:
I will limit the address bits to 24. Also 30 would work, but then you get JavaHeapSpace errors, if you don't modify the memory setting of the JVM.

On my system, if 32 data bits are used also 8 characters are shown.
And the number of characters is calculated by (bits-1)/4+1, which seems rock solid.

It could be, that your screen is to small to show 16 columns with 8 characters each.

roy77, 2017-08-22:
OS: Windows 10 Home

Dialog after pressing "Edit"
[image]

hneemann, 2017-08-22:
So your problem is the default window width of the data editor dialog?
You don't want to increase the window width manually?

roy77, 2017-08-22:
So simple... Sorry, I was in "Excel" mode. Double click on the columns border, right mouse for context menu...

hneemann, 2017-08-22:
Never mind!

closed 2017-08-22 by hneemann