hneemann/Digital #47: RAM Data dialog not updated after restart

roy77, 2017-08-22:
In "Measurments" dialog it is possible to open "Data" dialog which shows the content of a memory. The content is correct updated during simulation. When simulation is stopped the dialog stays opened. After starting the simulation again the content is not updated anymore. The dialog has to be closed and reopened.

hneemann, 2017-08-22:
This behavior is intended: It allows  to compare the RAM contents of different simulation runs.

You can start the simulation, open and keep the RAM data dialog, modify something in the circuit, start again, open a second RAM data dialog and compare both.

roy77, 2017-08-22:
What do you think about changing the textcolor to gray or changing the title when dialog data is not actual anymore?

What do you think about allowing to save the current dialogs information (position, size, state, slider position) and in next simulation start with the same configuration.

hneemann, 2017-08-22:
It's easy to indicate that the data is outdated. But its very hard to reopen all opened data dialogs if the model had restarted.

What do you think about the variant implemented in "v0.13-119-gd80217d-dirty" which you can find [here](https://infdigital.dhbw-mosbach.de/).

roy77, 2017-08-22:
Sorry, I do not see any differences to the previous behavior.

[image]

hneemann, 2017-08-22:
If the simulation stops the table contend should become blue.
And if the dialog is reopened, it should become visible at the position of the old dialog.
This works fine on my linux machine.

roy77, 2017-08-22:
The described behavior would be fine. Maybe it is not included in the linked file. I tested it in a Lubuntu VM with
- OpenJDK Runtime Environment (build 1.8.0_131-8u131-b11-2ubuntu1.16.04.3-b11)
- Build git-Revision: v0.13-119-gd80217d-dirty Build Time: 2017-08-22 16:51
and see nothing from the described behavior.

hneemann, 2017-08-22:
Hmm! I'm sure its included. But I just pushed the changes.
Please try HEAD.

hneemann, 2017-08-22:
I did not remember the second way to open the data dialog via the measurement dialog.
And that even though you mentioned it in your first sentence. :-)
[Now](https://infdigital.dhbw-mosbach.de/) it should be fine.

closed 2017-08-22 by hneemann