hneemann/Digital #53: Add persistent RW memory

Dexter9313, 2017-09-05:
I'm willing to build a computer with Digital but as far as I know it will not be able to write anything to disk because the only persistent memory we have is ROM.

Would it be possible to add a "hard drive disk" component tied to a specific file or something equivalent ? A nice start would simply be to be able to copy/paste the whole data of a RAM module to manually save or load the content of writable memory as Logisim allows for example.

hneemann, 2017-09-06:
Can you explain what you want to do with this component? How do you want to use it?
It seems to me that you need something like an EEPROM?
Something you can hook up to the data and address bus like a RAM?

Dexter9313, 2017-09-06:
I want to be able to write data in the component that will not be erased at the end of simulation. It can be seen as being persistent RAM. The aim is to have a storage component.
In my specific case I built a processor that I program using ROM but I can't save its output. Let's say I wrote an assembler in binary. My processor could use it to generate binary code from character strings I would feed it but I wouldn't be able to retrieve the result to reuse my program compiled by my processor.

For the use case, something that has the same interface and semantics as single or dual port RAM would be perfect, the only thing that it would have more than RAM is that its content persists throughout the different simulations (and Digital opening/closing). Being able to manually set its data with a data editor or saving/loading a file like for ROM would be a nice addition but is not mandatory. The priority is that my CPU can persist what it wants to persist.

So basically yes, an EEPROM would qualify for the job (even if its interface is not exactly the same as RAM because it is erasable, it is close enough). I think a larger user base would benefit from having an EEPROM than some "persistent RAM" as I described in the use case, because EEPROM is a standard component with a standard interface already. I don't want you to add anything to Digital that would confuse other people.

hneemann, 2017-09-06:
What I can do very easy is to add a save button to the RAM editor dialog which allows to store its content to a file. Then its possible to load the data to a ROM to run a generated binary.
Adding a persistent memory is something that I have to think about a little bit because there are several questions:
1. Where to store the files.
2. How to ensure that the file is found if circuits are moved around in the files system.
3. How to avoid confused users because the simulation of a circuit starts only once, because a memory is overwritten permanently, and the behavior has changed.
4. What about a reset of the memory?

Dexter9313, 2017-09-06:
A save button on RAM would be a nice temporary solution, yes ! Maybe a load button could also be useful to avoid having to load data from ROM to RAM (if writing on top of previously saved data is needed). It is not absolutly necessary though as we can load ROM data to RAM data anyway, it would just be convenient.

As for true persistent memory, I suggest these things off the top of my head : 
1. 2. You could store the data in the .dig file directly as you do for ROM (I guess you do). The user would have to save the circuit to make the memory it contains persistent but so is needed for ROM if edited. I don't know what the other users would think about it but I wouldn't mind.
3. 4. Making the memory resettable would help users initialize their circuit with the same initial state for every simulation, although I don't know why they would bother using persistent memory if they want it to be resetted for every simulation... They should just use RAM for that use case.

hneemann, 2017-09-06:
I have added an EEPROM.
In the circuit editor it behaves like a ROM, but you can write to it like to a RAM.
Up to now a programmatic modification of the content does not mark the circuit as modified.
You can play around with a [pre-release](https://infdigital.dhbw-mosbach.de/) build (v0.14-10).

Dexter9313, 2017-09-06:
I played with it a bit and it behaves exactly as I imagined.

Thanks a lot for the fast and efficient response !

closed 2017-09-06 by hneemann

hneemann, 2017-09-06:
added in v0.15