hneemann/Digital #54: Edit attributes when more than one component is selected

roy77, 2017-09-10:
Simple version
Allow editing attributes when selected components are from same type
- "Reset all diodes and FGETs" is not needed anymore. The selection allow a group wise reset and set
----------
Advanced version
Allow editing identical attributes when selected components are from different types. 
- fast changing of "Data bits" for all components

Maybe all attributes need internal parameter "allow group wise edit" to define which parameter support this feature (not for Label, Description ...)

hneemann, 2017-09-10:
Nice idea!
How to deal with values which are not the same in different components?
The usual way - leave the field blank - is very hard to implement.

roy77, 2017-09-10:
The only implemented solution I know is the blank field. 

Grouping would be a new approach.

Example:
Component 1 with Data with 12
Component 2 with Data with 4
Component 3 with Data with 12

Generate two rows with Data with. One with 12 and one with 4. Behind the value is a field with the grouped components label name.
Problem: Often the component name is not set.
Other solution. When a row is selected mark the corresponding group components in the circuit.

The marking approach enable another feature. Fast label settings. We define that the label is not groupable but visible. So each selected component label get a row in the attribute list. This allow fast labeling of all components.

I have no idea if this approach is easier to implement. It is the only I can think of.

hneemann, 2017-09-12:
This is the solution I like the most. What do you think?
[v0.14-25-g60c9898](https://infdigital.dhbw-mosbach.de/)

hneemann, 2017-09-12:
Added in v0.15

closed 2017-09-12 by hneemann

roy77, 2017-09-12:
I like the "Modify this value" solution. I think it would be better to disable/enable a component depending on "Modify this value".

hneemann, 2017-09-12:
Good idea. This makes the usage more clear.