hneemann/Digital #55: Usability

codenamecpp, 2017-10-16:
First of all, I'd like to say that it is great project and I like it very much. But i'm getting frustrated every time I used the program :(

It is pain to move wires and components around once I've place them. Single left click on component for moving it right away - it's very hard to get used to. When I try to select group of components that already wired up, it is nearly impossible without touching connected wires (for example, if I want to select and delete them, it can be done only individually for every single component). No shift + click combination to combine selection.

Single left click to place wire - it's not what I need in most cases. I am expecting Pointer tool be active by default (like in Logisim).

And one more thing: I cannot find how to enable grid dots in document, it difficult to me to wire thing up without grid.

Anywat, Digital is so awesome and intuitive after using Logisim, thank you!

hneemann, 2017-10-16:
_It is pain to move wires and components around once I've place them. Single left click on component for moving it right away - it's very hard to get used to. When I try to select group of components that already wired up, it is nearly impossible without touching connected wires (for example, if I want to select and delete them, it can be done only individually for every single component)._ 

See the discussion [here](https://github.com/hneemann/Digital/issues/6#issuecomment-304053838).

_No shift + click combination to combine selection._

Thats a feature which is hard to implement because up to now a selection is simply a rectangle given by two coordinates. So far I thought it's not worth the effort, but maybe I'm wrong. Hmm!

_Single left click to place wire - it's not what I need in most cases. I am expecting Pointer tool be active by default (like in Logisim)._

There is no distinction between the Pointer-Tool and the Edit-Tool in Digital because you can't modify a running simulation. Digital is either in edit-mode or in run-mode. Maybe I did not understand you correctly? 

_And one more thing: I cannot find how to enable grid dots in document, it difficult to me to wire thing up without grid._

See #56

codenamecpp, 2017-10-16:
Thanks for reply.

_>> Maybe I did not understand you correctly?_

What I am trying to say that now 2 actions is assigned on left click implicitly: Select and Wiring. Just a bit confusing.

One little thing I've forgot to mention: each time I select component in Component Tree View, mouse cursor jumps right - it looks strange, maybe add checkbox to disable this in settings?

hneemann, 2017-10-17:
_One little thing I've forgot to mention: each time I select component in Component Tree View, mouse cursor jumps right - it looks strange, maybe add checkbox to disable this in settings?_

I don't want to increase the number of settings to much. Maybe I should remove the mouse movement completely. I'm not sure. See #57

closed 2017-10-17 by hneemann

JarrettBillingsley, 2017-10-25:
I know this issue is closed but hopefully I can still give some input. I'm an instructor at university and I use Logisim for my computer architecture courses, but I've been wondering if there's something better...

>> No shift + click combination to combine selection.

> Thats a feature which is hard to implement because up to now a selection is simply a rectangle given by two coordinates. So far I thought it's not worth the effort, but maybe I'm wrong. Hmm!

This is a feature I use *constantly* in Logisim that I sorely miss. It is very difficult to select things with the rectangle in anything but the simplest circuits. I keep expecting the selection behavior to work like Logisim (and Powerpoint, and many other programs), where clicking on something selects it, but you have to click-drag to move it.