hneemann/Digital #56: Add a grid to the main panel

hneemann, 2017-10-16:
see #55

hneemann, 2017-10-16:
@codenamecpp
What do you think about the grid implemented in v0.14-52 which you can find [here](https://infdigital.dhbw-mosbach.de/).

codenamecpp, 2017-10-16:
Well, it seems OK for me, but one strange thing I noticed when zoom out, some of grid pixels change its color from black to light gray: https://gyazo.com/c1807a9f974a2d73ddafdf686cbc6721
Not sure if it was done on purpose...

hneemann, 2017-10-16:
The pixels don't change color. Due to rounding issues the dots get a different size.

codenamecpp, 2017-10-16:
Hm, looks like bug :)

hneemann, 2017-10-16:
I also think so! ;-)
But it seems to me that this is a problem somewhere deep buried in the swing libraries.

codenamecpp, 2017-10-16:
Unfortunately I not familiar with swing libs and I dont know how things being drawn in there. Something like creating shapes and putting them into surface and then perform some sort of transformations to scale/zoom? Is there way to add bottom layer for grid with no transformations at all and then draw top overlay with components and transformations?

hneemann, 2017-10-16:
Thats a good idea! I will think about it!

hneemann, 2017-10-17:
@codenamecpp 
Please try [v0.14-53-g7183ed5](https://infdigital.dhbw-mosbach.de/).

codenamecpp, 2017-10-17:
Almost perfect, but still there are some problems:
1. Noticeable gaps between points on zoom-out : https://gyazo.com/7fde827e76a50014d219a825593f4a9b
2. Dots shifting when resize window, maybe it because of rounding, see gif: https://gyazo.com/4f00b85fdb9b67ca82104ba59d7fa10a

I hope this problems could be fixed but even not, grid is good already.

hneemann, 2017-10-17:
added in v0.15

closed 2017-10-17 by hneemann