hneemann/Digital #59: Defining custom Java components

itoshkov, 2017-10-19:
It would be good to be able to extend the available components with custom ones. For example, one can add mouse or joystick inputs, floppy and hard disks backed by the host file system, network connectivity with the external world, etc.

hneemann, 2017-10-20:
I've already thought about implementing it.
But Digital is open source. You can already add components as you wish.
And with the available components also many examples are available.
So I think that it is not worth the effort to implement an appropriate interface.

itoshkov, 2017-10-20:
By "interface" do you mean API or UI?

If it's the former, I understand that it may be a bit premature to commit to it, until the system is more mature. But that can be added as a warning that this is a new and experimental feature, so people can decide if they want to invest their time right now.

If you mean UI, then I don't think you need much more than you already have. In Logisim there's just a simple menu action which loads a JAR file and optionally asks for the entry class if it's not defined in the JAR's manifest. I think you can even skip the latter part and just give error message in this case.

As for whether it's worth it, I think it is. As far as I know, Logisim is used by many courses, and some of them develop and use custom components. I guess many of these courses would love to have a newer and faster alternative, such as Digital, but they will need their components.

Yes, one can get the source and modify it to suit one's needs, but this has a couple of drawbacks:

1. They'll have to update and release the modified version with their custom code with new releases of Digital, or be left behind (more probable outcome)
1. The new versions of Digital can break the API without warning, so they'll have even less motivation to keep up
1. There may be multiple projects which provide custom components. This means that there are multiple custom versions of Digital. If I want to use two components from different "providers", I'll have to create a separate version of the system with both components.

So, while technically this is doable, I don't think it's a good solution.

P.S. If you think it makes sense, I can try and create some implementation of this feature.

hneemann, 2017-10-21:
I have added a simple jar library mechanism which allows to implement custom components in java. 
Here you can find a working [example](https://github.com/hneemann/digitalCustomComponents) containing a custom AND and OR component.

Let me know what you think about it.

hneemann, 2017-10-21:
added in v0.15

closed 2017-10-21 by hneemann

itoshkov, 2017-10-24:
I finally managed to play with this and it works! Thank you for your effort with this and with Digital in general.

Here are a couple of improvements you may consider for the future:

- Currently there is only one extensions file and it is set-up globally. It may be better to support these as component libraries, which we can add to different projects. This means that you'll have to put the libraries paths in the project file and load it more dynamically
- Publish the Digital.jar files to the central maven repository, so they can be easier to reference in the custom build scripts
- Make `Key.KeyInteger`, `Key.KeyBits` etc. constructors public, so that they can be used in custom components.