hneemann/Digital #61: Digital freezes after saving the diagram

smiszym, 2017-10-24:
Environment: Ubuntu 16.04 LTS
Version: v0.14, prebuilt JAR from https://github.com/hneemann/Digital/releases/tag/v0.14

Steps to reproduce:
1. Start Digital from the console:
`$ java -jar Digital.jar`
2. Put something on the diagram (I put Components > IO > Input) so that Save button becomes enabled.
3. Save the circuit in the default directory (in my case it was `/home/michael`)

Actual result: the program freezes right after. It stops responding to clicking with mouse on anything.

Additional information: the issue does not reproduce if I save the circuit under a different directory, like `/home/michael/Downloads`.

Can you reproduce this?

hneemann, 2017-10-24:
If you store a circuit Digital rebuilds the Components/Custom menu by scanning the folder where you stored the file and all its sub folders recursively.
Thus if you store a circuit in your home folder your complete home folder will be scanned, which can take some time.
I will implement a maximal recursion depth to fix that.

hneemann, 2017-10-24:
fixed in v0.15

commit 63b144b0 (2017-10-24): limits the number of files scanned during rebuild of the component library. fixes #61