hneemann/Digital #66: Better way of inputting values on multi-bit inputs

JarrettBillingsley, 2017-10-25:
It's very nice that I'm able to type in values for inputs (unlike poking 0s and 1s in Logisim), but it's clunky and slow. Right click, type in value, click apply, type in different value, click apply...

It would be great to be able to type the values right into the input, or even have up and down buttons for it, like a typical GUI spinner control, like this:

[image]

hneemann, 2017-10-25:
@JarrettBillingsley 
What do you think about the usage of cursor keys in the single value dialog?

JarrettBillingsley, 2017-10-25:
That helps!

JarrettBillingsley, 2017-10-25:
I also wanted to ask, is there a faster way to build than `mvn install`?

closed 2017-10-26 by hneemann

JarrettBillingsley, 2017-10-26:
Double-thanks!

hneemann, 2017-10-27:
I forgot to mention: The pdf documentation is created during the test execution. So if you skip the tests you also skip the creation of the pdf's. Thus your distribution zip will contain an outdated or no pdf documentation.

JarrettBillingsley, 2017-10-27:
Ohh, okay. Well that shouldn't be a problem for testing changes then :D

hneemann, 2017-10-27:
If you want to play around with the code to test something, it is much easier to import the pom.xml to you IDE and to build from the IDE. In this case you don't have to use maven at all.

commit 86ab14d1 (2017-10-26): added spinner to SingleValueDialog, fixes #66