hneemann/Digital #68: Does not ask to save changes when the program is exited (!!!)

JarrettBillingsley, 2017-10-25:
I don't know if this is a platform-specific issue, I'm on OSX.

If I make a new document, then place some components, trying to open another or make a new document will correctly ask if I want to save or discard my changes.

But if I exit (File > Quit), it doesn't! Those changes are lost forever.

hneemann, 2017-10-25:
I'm not sure what you are talking about! Digital has no "File->Quit" menu item!

JarrettBillingsley, 2017-10-25:
I think this is an OSX issue then:

[image]

That's not the File menu, I'm just so used to that being File... lol

hneemann, 2017-10-25:
Not sure what happens. There is a listener registered to the main window which is called if the user wants to close the main window. This listener then asks the user if he wants to save or discard his changes. I have no idea why this is not the case on OSX.
What happens if you close the window using the window decoration button.

JarrettBillingsley, 2017-10-25:
Closing the window with the red close button DOES give the "save/discard" window. That's really strange. 

Apparently there's another event you have to listen for: 

https://stackoverflow.com/questions/10076104/java-swing-on-a-mac-listening-for-quit-event

hneemann, 2017-10-25:
I don't want to add some strange apple specific code to Digitals code base. 
But the more serious issue is that I can't test such code because I don't use apple hardware and also don't have easy access to OSX devices.

JarrettBillingsley, 2017-10-25:
That's understandable. I think it's easy to check the platform at runtime
and only do it on one OS though.

If I submit a pull request, do you think you can test it on a non mac and
see if it works?

JarrettBillingsley, 2017-10-26:
So, the diff is simple, but I can understand if you don't want to force a dependency on Java 9. 

This should work for other platforms as well (perhaps on Windows or Linux a similar thing can happen if you right click the icon in the taskbar and exit). 

```diff
--- a/src/main/java/de/neemann/digital/gui/Main.java
+++ b/src/main/java/de/neemann/digital/gui/Main.java
@@ -55,6 +55,8 @@ import javax.swing.filechooser.FileNameExtensionFilter;
 import java.awt.*;
 import java.awt.datatransfer.Clipboard;
 import java.awt.datatransfer.DataFlavor;
+import java.awt.Desktop;
+import java.awt.desktop.QuitStrategy;
 import java.awt.event.*;
 import java.io.File;
 import java.io.FileOutputStream;
@@ -151,6 +153,7 @@ public final class Main extends JFrame implements ClosingWindowListener.ConfirmS
     private Main(MainBuilder builder) {
         super(Lang.get("digital"));
         setDefaultCloseOperation(DO_NOTHING_ON_CLOSE);
+
         setIconImages(IconCreator.createImages("icon32.png", "icon64.png", "icon128.png"));

         keepPrefMainFile = builder.keepPrefMainFile;
@@ -1503,6 +1506,14 @@ public final class Main extends JFrame implements ClosingWindowListener.ConfirmS
         URL.setURLStreamHandlerFactory(ElementHelpDialog.createURLStreamHandlerFactory());
         FormatToExpression.setDefaultFormat(Settings.getInstance().get(Keys.SETTINGS_EXPRESSION_FORMAT));

+        if (Desktop.isDesktopSupported()) {
+            Desktop desktop = Desktop.getDesktop();
+
+            if (desktop.isSupported(Desktop.Action.APP_QUIT_STRATEGY)) {
+                desktop.setQuitStrategy(QuitStrategy.CLOSE_ALL_WINDOWS);
+            }
+        }
+
         MainBuilder builder = new MainBuilder();
         for (String s : args) {
             if (s.equals("experimental")) experimental = true;
```

hneemann, 2017-11-01:
Can you please try `v0.15-1-g25ed066-dirty` which you can find [here](https://infdigital.dhbw-mosbach.de/)?
Should run with Java 9 and Java 8 and it should avoid the "Quit"-problems when started with Java 9.

hneemann, 2017-11-11:
The code for accessing the Java 9 API from the Java 8 code using reflection is so ugly that I decided not to add it to the code base.

closed 2017-11-11 by hneemann