hneemann/Digital #70: Should be able to change handedness of splitters.

peterb12, 2017-11-01:
Splitters can be rotated in any of the 4 cardinal directions.  This is good!  But consider the attached image, where the user intent is to send wire 3 to the top-most component and the wire 0 to the bottom-most:

[image]

Ideally the user would be able to preserve the right-facing-ness of the splitter but reorder the direction of the outputs.  Logisim allows you to specify which wires in the splitter get which bits.  As near as I can tell, Digital does not yet. 

(You can actually accomplish what I'm talking about above by rotating the splitter and the inverting the input and output splitting fan-ins and outs, but that's a bit oblique and is not immediately obvious.)

peterb12, 2017-11-01:
EDIT: I just realized that you can also use the output splitting syntax to do this; so in my example, one could write "3-3,2-2,1-1,0-0" and get the desired behavior.  But writing "3,2,1,0" is disastrous.

I think the actual change I'm asking for is not in code, but in documentation: I find the paragraph explaining the syntax for the splitting to be confusing.

closed 2017-11-01 by hneemann

hneemann, 2017-11-01:
By the way: what did you define in the And circuit? 
Why not using the built-in And?

peterb12, 2017-11-01:
Solely as a way of learning the tool - I was essentially going through Chapter 1 of nand2tetris and re-implementing everything 'from scratch', with parts built entirely on NAND.

[image]

(the "Not" is similarly defined as a nand with one input connected to both inputs.)

hneemann, 2017-11-01:
Ok, thats fine! I was afraid you would build the gates with single cmos transistors.
Although this is possible, the simulation becomes very slow if hundreds of individual transistors are used. :-)

itoshkov, 2017-11-01:
@peterb12 nand2tetris is an amazing experience. Go for it!

I just want to point out that even with its native tools, you start with implementing everything with NAND-gates, but then you are encouraged to use the embedded implementations for larger components because it's much faster. So if you want to implement a full HACK computer in Digital, it's perfectly OK to use the built-in chips.

And now for the shameless plug: If you go this path, you will need to have a compatible display, for which I recently created a custom chip. You can find it here: https://github.com/itoshkov/digital-n2t.

commit 96fec7b4 (2017-11-01): Better description of splitter configuration. fixes #70