hneemann/Digital #71: Spinner in mulit-bit input

roy77, 2017-11-02:
The spinner changes the circuit value immediately. The other controls only after pressing "Apply". Uniform behavior would be nice.  I like the immediately changing, but in rare cases it is not desired. Maybe a checkbox to switch between immediately and only after pressing "Apply" can solve this. Default: immediately.

Pressing spinner up at 0 results in 0xffffffffffffffff. The value should depend on the current bit width.

hneemann, 2017-11-02:
> The spinner changes the circuit value immediately. The other controls only after pressing "Apply". Uniform behavior would be nice. I like the immediately changing, but in rare cases it is not [...]

You are right: The behavior is not uniform:

1. If a value is added by the keyboard and enter is pressed, the value is set and the window is closed.
2. If you use the spinner the value is set immediately but the window is not closed.
3. If individual bits are set, not all can be set at the same time. It therefore makes sense to use the value only when "Apply" is pressed.

I don't know how to make this uniform. I don't want to add more checkboxes or buttons. That's also more confusing rather than helpful. Hmm!

> Pressing spinner down at 0 results in 0xffffffffffffffff. The value should depend on the current bit width.

Thats a good point. I will fix that.

hneemann, 2017-11-04:
After thinking about the dialog, I want to leave it as it is. It's not consistent, but I think it's quite usable.

closed 2017-11-04 by hneemann