hneemann/Digital #76: Bit extender?

22459, 2017-11-10:
Hi,

is there a reason Digital doesn't have the bit extender?: [http://www.cburch.com/logisim/docs/2.5.0/en/libs/base/extender.html](http://www.cburch.com/logisim/docs/2.5.0/en/libs/base/extender.html) or did I just not find it in the component tree?

Thanks

hneemann, 2017-11-10:
There is no reason. A bit extension is so simple that I thought a special component was unnecessary.
A bit extension is not a special logic. It's just a special kind of wiring:

[image]

22459, 2017-11-10:
Oh, I figured I should do something like that but couldn't figure out how. That makes sense. :)
Thanks.

closed 2017-11-10 by hneemann

hneemann, 2017-11-10:
@22459 
If you want to use the new bit extender (which only works with signed values) you can use the last [pre release version](https://infdigital.dhbw-mosbach.de/).

commit 5c0a9a07 (2017-11-10): updated release notes, fixes #76