hneemann/Digital #77: 32 bit components

ggalfi, 2017-11-11:
Hi,

first of all I'd like to echo the kudos for this great project. It is really fun to play with.
I begun to build a 32bit RISCV processor in the simulator so certainly I need 32 (or more) bits to operate on. There are practically two issues related to 32 bit components:

1. When using the shifter component for right arithmetic shift I got erroneous results for 32 or more bits. This problem could be easily corrected by changing the line
int mask = (1 << (bits)) - 1;
to 
long mask = (1L << (bits)) - 1;
in
BarrelShifter.java

2. This is not a bug rather some convenience issue: I run into error messages when I try to define 32bit unsigned constants (especially for the value 0xFFFFFFFF). The root cause of this problem is the usage of Integer.decode method. I changed the definition for VALUE key definition to make it capable to handle long values. For this I have to do minor changes in Key.java, Keys.java, EditorFactory.java, CircuitBuilder.java. I can provide you with the diff file of the changes if you want to incorporate these in to the main branch. Some care will be needed here because with my changes in the .dig files "<long>" tag should be used instead of "<int>". For me it wasn't a problem because I manually adjusted these tags in my dig files, but obviously it is not an option for every dig file.

With the changes mentioned above my project went on smoothly, I implemented almost the entire RV32I instruction set, even in its current state I can compile some C codes to run them on my simulated machine.

Cheers,
Gergely

hneemann, 2017-11-11:
This is very interesting. I also thought about a RISC V implementation, but then went for a simpler processor design that I created myself. Are you planning to publish your implementation?

I will think about the 32 bit constants. Maybe its possible to fix this problem without breaking old files.

hneemann, 2017-11-11:
Please try the pre-release version v0.15-38-g563dfe5 which you can find [here](https://infdigital.dhbw-mosbach.de/).

ggalfi, 2017-11-11:
The RISC V specification looks a bit complicated at first glance, but it surprised me that only a relatively few component was needed to implement it (I expected much more complexity when I've begun the project). Surely the guys who had designed the spec had  the main goal to keep the hardware implementation as simple as practically possible.
I am twiddling with this RISC V stuff just for fun, so don't have any plan what to do with it. If you'd consider to incorporate it in the Digital source tree as an example, I'd be happy to contribute.

ggalfi, 2017-11-11:
I've tested my stuff on this v0.15-38-g563dfe5 version, and both the right arithmetic shift and my 32 bit constants are working as they should be. From my point of view this issue could be closed. Thanks for the quick fix!

commit 4119ed81 (2017-11-11): updated release notes, fixes #77
closed 2017-11-11 by ggalfi