hneemann/Digital #79: fix pin error

hydrogenbomb1212, 2017-11-12:
if i have a input pin name in1 with pin number 1
and a output called out1 with pin number 3
i get a error telling me there is a error in my subcircuit - couldn't resolve pin layout in (subcircuit)

i got the error when i used 2 2-bit registers to make a 4-bit register
it fixed when i adjusted the pin names to:
clk
in 1
in2
out3
ou4

original names:
clk
in1
in2
out1
out2

hydrogenbomb1212, 2017-11-12:
i know i can use premade registers, i just want to use a minimum of premade objects in my circuit.

hneemann, 2017-11-12:
I do not understand what your problem is. Could you provide a sample circuit? You can simply attach a ZIP file to the issue.

But there was an error that causes a problem when an input pin is used in a sub-circuit that has no connection in the sub-circuit (see #74). This bug is fixed but not yet released. You can try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

hydrogenbomb1212, 2017-11-13:
well, the problem dint repeat, ill notify you if it happens again

hneemann, 2017-11-13:
seems to be a duplicate of #74

closed 2017-11-13 by hneemann