hneemann/Digital #80: TCP communication

roy77, 2017-11-16:
Starting the simulation with start button in Digital do not change the measurement dialog, but "start" command closes the measurement dialog. 

"debug" always start with disabled real time clock
"start" only start with enabled real time clock if real time clock is really enabled in dig

Why not allow switching between debug/start without reset all memories and registers? So "debug" appears as pause. If I want a reset I can use "stop".

hneemann, 2017-11-16:
> Starting the simulation with start button in Digital do not change the measurement dialog, but "start" command closes the measurement dialog.

I will fix that.

> 
> debug" always start with disabled real time clock
> "start" only start with enabled real time clock if real time clock is really enabled in dig
> 
> Why not allow switching between debug/start without reset all memories and registers? So "debug" appears as pause. If I want a reset I can use "stop".

In the moment it is not possible to switch between real time clock running or not in a running simulation.
This is because if the real time clock is running, the model is executed in its own thread and only the GUI updates are done in the GUI thread. 
If the real time clock is not running, the model runs in the GUI thread because the only modification to the model happens by the GUI, so a model-thread makes no sense.

This behavior is perfectly fine if a simple circuit is simulated or a small program is executed in a simulated processor. 
But it does not allow to start a program in debug mode, click "run to break" and then interact with the GUI of the simulator, while the program is running. You are only able to interact with the model if the processor is stopped at a break point. This problem is also mentioned in #62, i think.

I'm thinking about changing that, but that's not easy.

roy77, 2017-11-16:
One solution could be to return state (register, counter, memory, etc.) as structure when stopping and to enable the state to be set at startup. Then the worker thread could be stopped and restarted as a GUI thread with the state.

hneemann, 2017-11-16:
In the moment Digital doesn't have something like a separate state for the components.
This is not necessary because the simulation model is completely flat. All components in the sub circuits are present as often as they are used. So it is not necessary to handle a state separate from the component: The components store their state directly. To add such a state is a huge amount of work and adds lot of code.

hneemann, 2017-11-16:
See #83

closed 2017-11-16 by hneemann