hneemann/Digital #86: Fast run?

itoshkov, 2017-11-24:
The [README](https://github.com/hneemann/Digital/blob/master/README.md) says that there is a "fast-run" mode without updating the GUI. But I couldn't find any more information about it, so I dug up in the code. What I found is this [ThreadRunner](https://github.com/hneemann/Digital/blob/a83f056c585fb002dce208f6e3567bdc26a8a68e/src/main/java/de/neemann/digital/draw/model/RealTimeClock.java#L85), which is used when the frequency is higher than 50000 Hz.

However, when I run a simulation with it Digital still updates the values of various buses etc. Is it possible to run the simulation without _any_ such updates?

hneemann, 2017-11-24:
The fast run mode is only available if a BREAK component is present in the circuit.
In this case the fast run icon is enabled. Pressing this button the model is clocked as fast as possible without a GUI interaction until a rising edge at the input of the BREAK component is detected.
The idea is to add a BRK assembly instruction to the processors instruction set which creates a rising edge at a BREAK component if executed. In this case the fast run mode becomes a "run to break".

itoshkov, 2017-11-24:
I added a break, but I had to dig a little more to see how to enable this. Turns out that I have to also unset the "Start real time clock", which I previously have set. And then I need to first "Start the simulation" and after that I had fast run enabled.

But then it gave me an error: "Break timeout after 100,000 cycles".

itoshkov, 2017-11-24:
I found that the number of cycles can be changed by the BREAK component's settings. I made it to be 10,000,000, but then Digital didn't show the graphics component until I hit this timeout.

What I'm actually trying to do is to run the computer that I've created, so I can try and play a game on it. When I run it with the real-time clock, the speed isn't good enough even when I choose 500,000 Hz, so I wanted to check with this fast run.

I can understand if Digital cannot actually run the thing fast enough. I just want to see how fast it can go.

hneemann, 2017-11-24:
You are right: The real time clock needs to be turned off. I forgot to mention that.

In this mode there is no interaction with the GUI. You can easily run into a state where the fast run mode never stops,  because no break signal is generated. To avoid that, there is a "time out" cycle count in the break element. If this cycle count is reached the fast run mode is stopped and an error message shows up.

hneemann, 2017-11-24:
I'm not sure the fast run mode can improve the overall speed of the simulation compared to a clock which is set to the fastest possible value. If a frequency above 50kHz is selected the actual frequency setting is ignored. Instead the simulation of the model runs in it's own thread, as fast as possible. But certain events can cause a repaint of the GUI. This repaint happens in the GUI thread but is synchronized with the model execution thread to avoid inconsistent drawings by updating the model while it is being drawn. If you want to check out how it looks like without this synchronization you can remove the sync from the repaint by removing  the lines 203 and 213 in the file de/neemann/digital/draw/elements/Circuit.java.

If you don't want to remove the synchronization you should put your computer in a embedded circuit so that the main circuit is as simple to draw as possible. Doing so, you get the fastest possible speed.

itoshkov, 2017-11-24:
Thank you, I'll try that.

hneemann, 2017-11-24:
Maybe a more elegant solution is to export your design to VHDL and run it on an FPGA. In this case it should be possible to reach 20MHz or more.  I have recently tested my own processor design running at 25MHz and it works just fine.

hneemann, 2017-11-25:
Is it possible to export your processor to VHDL if you remove the hack display from the circuit by replacing it by a simple RAM?

If this is possible without creating error messages during export i can provide a hack display simulation which creates VGA compatible signals if running on an FPGA. Thus you can connect a real monitor to the FPGA to show the graphic output on that monitor.

The FPGA [evaluation board](https://www.digikey.com/product-detail/en/digilent-inc/410-183/1286-1041-ND/4970275) I use also provides 5 push buttons. If you can modify your game to be controlled by 5 digital inputs, one can play the game running on the eval board. :smile:

closed 2017-11-25 by hneemann

hneemann, 2017-11-26:
I have fixed a weakness in the event dispatch system that has annoyed me for a long time.
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) to get about 10% speed improvement. That's all you can get. :smile:

itoshkov, 2017-11-26:
Thank you again. I'll take the 10% improvement any day :)

I just wanted to see if this can be used to run the HACK computer in good speed. It is far faster than what Logisim can do and is comparable with the CPU emulator provided with the course.

In the course one builds a higher abstraction Virtual Machine and the course provides a VM Emulator, which is much faster than the CPU one. So overall Digital is pretty impressive as it is!

I'm generally interested in FPGAs, but looking at my schedule and my total inexperience with building _real_ hardware, I'm not sure when I'll be able to get into it.

hneemann, 2017-11-26:
I'm interested in hearing about the pull request I sent today. These changes may be noticeable. It depends on the running machine code.

You are right: _Building_ real hardware is really difficult. But buying a trainer board is just expensive, not difficult. 😄 
And VHDL and Verilog are a bit more complex than the n2t HDL, but not to much. At least not when you do simple things.

itoshkov, 2017-11-27:
I just tried it. Your 10% speed-up estimate seems about right. It takes about 25.78 seconds to show the display with the new code, vs. about 28.25 seconds with the previous commit. I measured the runs **very accurately** by pressing the stopwatch on my phone in exactly the right moments!

P.S. Stop tempting me! :) I remember how great it felt to see it working in an emulator, so I can only guess that it's going to be even better with the real thing. But this won't happen before February for sure.

hneemann, 2017-11-27:
Much more interesting is [this](https://github.com/itoshkov/digital-n2t/pull/3) pull request.
If you write a lot to your display this pull request can make a huge difference.
If you like, you can [send](mailto:digital-simulator@web.de) me your circuit. Then I'll see what's possible.