hneemann/Digital #87: Plus key accelerators do not work with English language keyboards

xotmatrix, 2017-11-27:
On English language keyboards, because of the + key is really a shifted = key, a plus keypress is not detected. This affects both the zoom control and the input pin number control.

At the end of topic linked below there seems to be a fairly comprehensive solution which works with multiple keyboards and adds NumPad support as well.

https://community.oracle.com/thread/1356291

closed 2017-11-28 by hneemann

hneemann, 2017-11-28:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) version  and let me know how it works on your machine.

xotmatrix, 2017-11-28:
This fix works somewhat. It does require holding the shift key for + which I think is rather awkward. Applications that use +/- in this sort of way rarely if ever require that of English language keyboard users; the unshifted = key is treated like the + key of other keyboards. That's in part why I linked that particular solution above.

hneemann, 2017-11-28:
So the common way to handle this is: Name it [+] but use [=] instead? :-)
You will find that behavior in the latests pre release.

xotmatrix, 2017-11-28:
Perfect! Thank you.

commit 3cdfc017 (2017-11-28): fixes #87