hneemann/Digital #91: HTML text component

roy77, 2017-12-19:
The new grid function makes reading text components more difficult. This can be solves by drawing a white rectangle behind text components.
A more flexible solution is to allow html in text components.

With JLabeI I get good results, but SVG and Latex support is not implemented. Also it is not compatible to the old "\n" version because of the changes in Textshape.java. Without the Textshape.java changes the complete html code have to be written in one line.

[image]

GraphicSwing.java
`

        public void drawText(Vector p1, Vector p2, String text, Orientation orientation, Style style) {
        JLabel lbl = new JLabel(text);
        applyStyle(style); // sets also font size!
        int height = gr.getFontMetrics().getHeight();
        if (height > minFontSize) {
            if (text == null || text.length() == 0) return;
            //GraphicMinMax.approxTextSize(this, p1, p2, text, orientation, style);

            text = cleanLabel(text);

            boolean rotateText = false;
            if (p1.y == p2.y) {   // 0 and 180 deg
                if (p1.x > p2.x)   // 180
                    orientation = orientation.rot(2);
            } else {
                if (p1.y < p2.y) // 270
                    orientation = orientation.rot(2);
                else            // 90
                    orientation = orientation.rot(0);
                rotateText = true;
            }

            AffineTransform old = null;
            old = gr.getTransform();
            if (rotateText) {
                old = gr.getTransform();
                gr.translate(p1.x, p1.y);
                gr.rotate(-Math.PI / 2);
                gr.translate(-p1.x, -p1.y);
            }

            int xoff = 0;
            if (orientation.getX() != 0) {
                int width = gr.getFontMetrics().stringWidth(text);
                xoff -= width * orientation.getX() / 2;
            }

            int yoff = 0;
            if (orientation.getY() != 0) {
                yoff += height * orientation.getY() / 3;
            }
            gr.setColor(Color.black);
            Dimension size = lbl.getPreferredSize();
            lbl.setBounds(0, 0, size.width, size.height);
            gr.translate(p1.x, p1.y);
            lbl.paint(gr);
            //gr.drawString(text, p1.x + xoff, p1.y + yoff);

           // if (rotateText)
                gr.setTransform(old);
        }
    }
`

GraphicMinMax.java
`

        public static void approxTextSize(Graphic gr, Vector p1, Vector p2, String text, Orientation orientation, Style style) {

        if (text != null && text.length() > 0) {
            JLabel lbl = new JLabel(text);
            Dimension dim = lbl.getPreferredSize();

            Vector delta = p2.sub(p1).norm128();
            Vector height = new Vector(delta.y, -delta.x).mul(dim.height).div(128);
            //            Vector height = new Vector(delta.y, -delta.x).mul(style.getFontSize()).div(128);

            int textWidth = dim.width; //getTextWidth(text, style);
            Vector width = delta.mul(textWidth).div(128);

            Vector p = p1;
            if (orientation.getX() != 0) {
                p = p.sub(width.mul(orientation.getX()).div(2));
            }

            if (orientation.getY() != 0) {
                p = p.sub(height.mul(orientation.getY()).div(2));
            } else
                p = p.sub(height.div(4));

            gr.drawPolygon(new Polygon(true)
                    .add(p)
                    .add(p.add(width))
                    .add(p.add(width).add(height))
                    .add(p.add(height)), Style.THIN);
        }
    }
`

Textshape.java        
`

        
    public void drawTo(Graphic graphic, Style highLight) {
        StringBuilder sb = new StringBuilder();
        Vector pos = new Vector(0, 0);
        /*
        final int dy = (Style.NORMAL_TEXT.getFontSize() * 20) / 16;
        for (int i = 0; i < text.length(); i++) {
            char c = text.charAt(i);
            if (c == '\n') {
                if (sb.length() > 0) {
                    graphic.drawText(pos, pos.add(1, 0), sb.toString(), Orientation.LEFTTOP, Style.NORMAL_TEXT);
                    sb.setLength(0);
                }
                pos = pos.add(0, dy);
            } else
                sb.append(c);
        }
        if (sb.length() > 0)
            graphic.drawText(pos, pos.add(1, 0), sb.toString(), Orientation.LEFTTOP, Style.NORMAL_TEXT);
            */   graphic.drawText(pos, pos.add(1, 0), text, Orientation.LEFTTOP, Style.NORMAL_TEXT);
         }
        }
`

hneemann, 2017-12-20:
If the grid disturbs the text its maybe better to modify the grid color. This also helps with all the other text fragments and not only with the text component.

roy77, 2017-12-20:
A other grid color is a good solution. In addition, adding HTML to the text component would be a simple solution for formatting text.

hneemann, 2017-12-20:
Are you more interested in color or size or both?

roy77, 2017-12-20:
I am more interested in size.

roy77, 2017-12-20:
A suitable solution for my intended use.

closed 2017-12-20 by hneemann