hneemann/Digital #92: Key "L" stops simulation

codexp, 2017-12-26:
v0.15.147

when pressing "L" key in simulation, it stops simulation and selects button input-element to place on circuit.

hneemann, 2017-12-27:
This behavior is intended. If you insert a new component, the simulation is stopped. The only thing one could do is to block component insertions while the simulation is running.

codexp, 2017-12-28:
this prevents using a button bound to L key in simulation.
I was confused when I've pressed L and simulation stopped.
But discovered that while designing the circuit, it is very helpful to have that key inserting last used component (very nice).

hneemann, 2017-12-28:
Maybe it's better to use CTRL+L to insert the last used component. Hmm!

codexp, 2017-12-28:
if you can manage a clear distinction of key layout between design mode and simulation, this may bring an advantage of using plain keys for editing features while designing (which is a little bit nicer than to press and hold modifier also).

commit d4f92724 (2017-12-31): Hotkey "L" is disabled if model is running. fixes #92
closed 2017-12-31 by hneemann