hneemann/Digital #93: IO: bidirectional port

codexp, 2017-12-26:
It would be nice to have a bidirectional port for sub-components to act as input and as output.

Why there is no 74LS245 component?

hneemann, 2017-12-27:
Every input or output is bidirectional by default.
There is no technical reason why there is no 74245. You can simply add such a IC.

codexp, 2017-12-27:
Ok, thank you!

I've tested it and indeed they are bidirectional.
It was kind of misleading to think that they do only what they are named after.

closed 2017-12-27 by codexp

hneemann, 2017-12-27:
The reason is in the structure of the simulation. All circuits are wired directly and all the inputs and outputs are removed and replaced by a direct connection. It is simillar to a real chip: The pins don't have a direction. Only the logic inside the chip has.

codexp, 2017-12-28:
So I suppose the distinction is only for IDE, for better readability?

hneemann, 2017-12-28:
You are right. And at the top most level there is a difference: You can define the value of an input and only display the value of an output.