hneemann/Digital #94: Feature Request: LogiSim import

davidsummers, 2018-01-02:
Your program looks great!

I started a couple of weeks ago on using LogiSim and it is pretty nice but I already ran into a few strange things and some bugs.

Would it be possible to import an already existing LogiSim circuit rather than having to re-draw it from scratch?

Or is there some way to do that already other than just selecting open?  I tried Digital "open" menu item and it didn't see any of my LogiSim circuits.

hneemann, 2018-01-02:
Digital is not able to import LogiSim circuits.
And it takes a lot of work to implement such a feature. I also think that the benefits of such a feature are minimal. Therefore, I do not think that there will be such an import in the near future. Sorry for that.

davidsummers, 2018-01-03:
OK, thanks, I was just hoping there might be some standard way to export from LogiSim and import to Digital.

I'm only a week into using LogiSim so will try to convert by re-drawing.

closed 2018-01-03 by hneemann