hneemann/Digital #95: Can't create 64-bit splitter

davidsummers, 2018-01-03:
I am a newbie on "Digital", maybe there is a better way?

I'm trying to create a bus with 64 bits that get split out or split (sometimes I need one way and sometimes the other).

I tried putting one in my circuit and I get the message "Error creatint the circuit, Input bits are defined several times!".

I tried several combinations:

Input Split       Output Split
====================
63, 1                 64
1, 63                 64

64                    63, 1
64                    1, 63

None of these seem to work and give me the above message.

Is this a bug?

Is there some other (or better) way to do that?

hneemann, 2018-01-03:
Thats a bug. :-(
Please try the latest [pre-release](https://infdigital.dhbw-mosbach.de/) build.

closed 2018-01-03 by hneemann

davidsummers, 2018-01-03:
Confirmed fixed!

Thanks a lot!

commit 61db5510 (2018-01-03): fixed a bug in the splitter consistency check, fixes #95