hneemann/Digital #96: Feedback

LarsiParsii, 2018-01-05:
First of all, thank you for making a simulator that has some functionality that Logisim didn't offer. My teacher recommended this software, and so far I'm enjoying it. I do, however have some feedback on the program:

- The EEPROM. It would be really helpful to be able to use binary instead of hex. I think there should be an option to change this.
- An option to change the default number format across the program. It can be quite tiresome to change the attributesof every register, or copy/pasting them.
- Being able to drag the wires as in Logisim.
- Being able to change the appearance of an embedded circuit (as in Logisim).
-Better representation of values. Again, a in Logisim, with the values inside a box while
- The attributes dialog box often shows up in the corner of the screen, so it has to be dragged to the middle to be able to see all the attributes.
- Being able to set a chip to "always on", when it's not supposed to be turned off.

Again, a great program that! :)

hneemann, 2018-01-05:
> The EEPROM. It would be really helpful to be able to use binary instead of hex. I think there should be an option to change this.

Binary values are very space consuming on screen. May it should at least be possible to enter binary values with the common notation 0b10111.

> An option to change the default number format across the program. It can be quite tiresome to change the attributesof every register, or copy/pasting them.

It's possible to select the components and change all the attributes at once.

> Being able to drag the wires as in Logisim.

See [this](https://github.com/hneemann/Digital/issues/6#issuecomment-304147130) discussion.

> Being able to change the appearance of an embedded circuit (as in Logisim).

See #39 

> The attributes dialog box often shows up in the corner of the screen, so it has to be dragged to the middle to be able to see all the attributes.

The attributes dialog should pop up at the actual mouse position. Can you explain in more detail what happens in your case?

> Being able to set a chip to "always on", when it's not supposed to be turned off.

Please explain in more detail what "always on" means.

LarsiParsii, 2018-01-05:
> The attributes dialog should pop up at the actual mouse position. Can you explain in more detail what happens in your case?
>
If I have a component in the corner of my screen, that will make the attributes box pop up in the corner, not showing all the options. Would be nice if the whole window could show when opened near a corner.

>Please explain in more detail what "always on" means.
>
I mean that the "enabled" pin is on on by default, without the need of an input.

hneemann, 2018-01-05:
> If I have a component in the corner of my screen, that will make the attributes box pop up in the corner, not showing all the options. Would be nice if the whole window could show when opened near a corner.

Are you using a windows machine? On my Linux machine the AttributeDialog is always complete visible. Hmm!

> I mean that the "enabled" pin is on on by default, without the need of an input.

See #36

LarsiParsii, 2018-01-05:
>Are you using a windows machine? On my Linux machine the AttributeDialog is always complete visible. Hmm!
>
Yes, I am using a Windows machine (10 x64).

hneemann, 2018-01-05:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) version.
Now it is possible to enter binary (0b10011) values in the DataEdtitor and the AttributeDialog should not go of screen anymore.

LarsiParsii, 2018-01-05:
Thank you for your quick support!
by the way, Would it be possible in the future for the user to change the EEPROM layout so that the addresses (in hex or binary) were all shown on a list vertically? It would make it a little easier to understand for students like me (however, this isn't essential).

Also, how can you supply an EEPROM with one wire for both input and output? I've connected them as shown in the picture, but no data is neither being inputted or outputted into the EEPROM.
I have this question as I've tried multiple times in Logisim to follow along Crash Course: Computer Science, but I find i hard to adapt their examples into Logisim/Digital. i do not expect you to answer this, but it would really help me out!
[https://www.youtube.com/watch?v=FZGugFqdr60](url)
[image]

LarsiParsii, 2018-01-05:
Tried using a relay instead, but to no avail. Seems like the EEPROM doesn't accept any input. Hmm..
[image]

hneemann, 2018-01-05:
Don't use Relays! They are very expensive to simulate. Use the three-state drivers instead.
If you put the data value on the input "D" and "CS" is high while "WE" goes high then the value is stored in the EEPROM. (Its possible to look into a RAM or EEPROM while the simulation is running by clicking on the chip.)

Maybe you should do your first experiments with a Harvard architecture, which means storing the program in a ROM chip and only store the data in a RAM chip. Then you don't need an EEPROM or an instruction register and the control unit is much simpler because it's stateless.

You can take a look at the processor example included in the distributed ZIP file. Or look at the MIPS architecture as an other example.

hneemann, 2018-01-05:
Binary values (0b10011) in the data editor and an attribute dialog that is always completely visible is included in v0.17.

closed 2018-01-05 by hneemann