hneemann/Digital #97: Feature Request: Option for register asynchronous clear

davidsummers, 2018-01-05:
The Register component doesn't currently have a way to asynchronously clear the value.

Could that be implemented?

I'm working on a Program Counter and it would be very nice to be able to have a Reset signal clear the Program Counter to 0 no matter what the state of the Clock signal is.

A Clear, either as an option or always there would be very helpful.

Thanks for a great program.

hneemann, 2018-01-05:
Do you need an enable input? If not then you can use a D-FF with asynchronous clear. D-FFs are also able to store more than one bit.

davidsummers, 2018-01-05:
Good point, I hadn't investigated all the components thoroughly.

I can do this by using a D-FF and just adding an AND GATE on the clock input.

Thanks, I'll use that instead for now.

hneemann, 2018-01-05:
It's not a good idea to use a clock signal which has passed a logic gate. It's better to use a muxer in front of the D-FF and simply store the value which is already stored if not enabled.

hneemann, 2018-01-06:
Here is an example of two identical state machines (a 3-bit counter). The only difference is, that the right one has a delay in the clock signal of one of the flip-flops.

[image]

closed 2018-01-06 by hneemann

davidsummers, 2018-01-07:
Wow, I hadn't seen that behaviour yet but I'm sure I will as I do more.

Thanks for the expert advice.

Very interesting.