hneemann/Digital #99: ATF1502AS TQFP support

roy77, 2018-01-08:
I bought an ATF15XX-DK3 to demonstrate simple circuits quickly. Unfortunately I did not pay attention that TQFP44 not working. Unfortunately, it is only in the manual and not in the readme that (only) PLCC is supported. In order for this to run with digital, the following enhancements would be necessary.

new Exporter with 
getPinMapping ()
                . setAvailBidirectional (2,3,5,5,6,8,10,11,12,13,14,15,15,18,19,20,21,22,23,25,27,28,30,31,32,33,34,42,43,44)
                . setClockPin (37);

Wincupl shows me "f1502isptqfp44" as identifier, but it doesn't work ("Wrong device name, Defaulting to PLCC44"). It works when I start the fitter with -dev P1502T44. The "DEVICE" entry seems to be ignored. With -dev P1502C44 the PLCC44 version is created, independent from the "DEVICE" entry.

It is possible to add a new exporter to the code or to define a new exporter which read his configuration from a file?

hneemann, 2018-01-08:
I did not know this board at all. And I was also unaware of the different pin assignments.
I think all these chips should be supported.
I'm not sure if a configuration file is the best solution because the number of chips is limited and does not change anymore.
Let me think about it.

roy77, 2018-01-08:
I don't know abel. I suspect with configurable settings (extra file, operator, Digital settings) for
-IO-pins
-Device name
-fitter name
-fitter command with a filename symbol 
it is possible to support all the packages of ATF1500 to ATF1508.

hneemann, 2018-01-08:
I've refactored the ATF150x code so it should be easier to add new devices.
Not sure that it still works. Tomorrow I will test it in the lab.

roy77, 2018-01-09:
Now it works with my board. 👍

hneemann, 2018-01-09:
PLCC package is also still working.

closed 2018-01-09 by hneemann