hneemann/Digital #100: fitter error on windows

roy77, 2018-01-09:
I am using Windows 10.

After further testing with different paths I get an error with fit1502.exe.
I get the error message Checking Extra wiring failed ...

I was able to reconstruct the problem from the command line.

D:\AtmelATF15xx>"C:\Program Files (x86)\Logisim\Fitters\fit1502.exe" test.tt2
Atmel ATF1502AS Fitter Version 1.8.7.8 (02-05-03)
Copyright 1999,2000 Atmel Corporation
Checking Extra wiring failed ...

Changing the current path to the fitter path, solve the problem

C:\Program Files (x86)\Logisim\Fitters>fit1502.exe  D:\AtmelATF15xx\test.tt2
Atmel ATF1502AS Fitter Version 1.8.7.8 (02-05-03)
Copyright 1999,2000 Atmel Corporation
---------------------------------------------------------
 Fitter_Pass 1, Preassign = KEEP, LOGIC_DOUBLING : OFF
 ...
 Design fits successfully

Please change the fitter start folder and parameter.

hneemann, 2018-01-09:
Keep in mind: These tools are almost 20 years old! :-)
I did not even try to install WINCUPL in a folder other than the default one.
Especially not in a folder which contains a space. 
Did you try to save the tt2 file in a folder that contains a space?

roy77, 2018-01-09:
I find the problem. It works when atmel.std is in current folder. I install and remove WINCUPL to get the fitter files and copy them to my Digital files. Maybe the remove of WINCUPL deletes the fitter path from my windows environmental variables. For me as user, the fitter is a part of Digital and want to give my students a zip with all necessary parts. Is it possible to set the fitter folder as the current folder when running the fitter to solve this? Otherwise I just put the tt2 files in the fitter folder.

hneemann, 2018-01-10:
I also have tested the calling of the fitters with the current path set to the path of the fitter instead of setting the current path to the path of the tt2 file.
In this case, no white space can be used in the full path of the tt2 file. However, you don't need to install WinCupl. All you have to do is copy the fitters into your file system. A big advantage.

Have you found a way to handle spaces in the tt2 path?

If not, I can simply add a check and generate an error message if a path containing spaces is used.

roy77, 2018-01-10:
You are right I missed that problem.I didn't find a solution for spaces in the arguments of the Fitters, but there is still a solution which allows spaces in the tt2 path and in the Fitter folder. The Fitter uses the environment variable "FITTERDIR" (created by Wincupl). If this variable is set the fitter find the needed files.
The following code creates the environment variable temporarily (only for procesBuilder)

OSExecute.java
public void addtoEnviroment(String variable, File Dir) {
    Map<String, String> env = procesBuilder.environment();
    env.put(variable, Dir.getPath());
    //        System.err.println(env.values());
}

StartATF150xFitter.java
execute.setWorkingDir(file.getParentFile());
execute.addtoEnviroment("FITTERDIR", fitterExe.getParentFile());

The put method create a new key or overrides the key if it exists. 
I did not work with Linux/Wine, so I can only say it works for windows.

hneemann, 2018-01-10:
I have added the environment variable. 
This variant still works with Linux, even if the TT2 path contains a space.

closed 2018-01-11 by hneemann