hneemann/Digital #102: Create curcuit for ATF150X with constants

roy77, 2018-01-11:
For my first test of ATF150X creation I design a binary to BCD converter. Therefore I need two constants. I get the error.

de.neemann.digital.builder.jedec.FuseMapFillerException: Constant values are not allowed!
        at de.neemann.digital.builder.tt2.TT2Exporter$ProdInput.add(TT2Exporter.java:355)

I solved this by using a XOR with same signal on both inputs to get a 0-singal. With a inverter I get a 1-signal. Together with splitter components I build my values. I do not like this workaround but the error disappears. I will test it on my hardware tomorrow.

But one problem remains. I want to set an output to a constant value. Therefore I use the XOR concept again but it doesn´t work, maybe the xor is removed during optimization.

[Test2.zip](https://github.com/hneemann/Digital/files/1623599/Test2.zip)

What is the reason for prohibiting constants?

hneemann, 2018-01-11:
In the current TT2 export implementation, it is quite complicated to set an output pin to a constant value.
And in almost all cases it is not useful to use a CPLD to create a pin with a fixed high or low value.
You can simply use GND or VDD instead.
As a work around you can use one of the switches and keep the switch in low or high state.
And by the way: Why is a constant output value needed in a BCD decoder?

roy77, 2018-01-12:
I am using the Double-Dabble Binary-to-BCD Conversion algorithm.

Attached the implementation for 7 bit binary (with my ATF1502 wit 44 pins only two of the four 7 segment are accessible). It´s the version with a lucid structure. For ATF1502 implementation I have to replace counter and constants with far more confusing structures.

[Bin2LED7.zip](https://github.com/hneemann/Digital/files/1625191/Bin2LED7.zip)

I know there exists a simple non-recursive Binary-to-BCD Conversion algorithm, I will switch to this.

hneemann, 2018-01-12:
But there is no constant output in your design. Maybe there was a misunderstanding: it is of course possible to use constants somewhere in the circuit. It is not allowed to create a circuit that causes an output pin to be constant.
And a counter can simply be replaced by a flip-flop and an adder. 

The problem with your circuit is that it uses a lot of macro cells. As far as I can see, the circuit uses 8 + 15 + 3 = 26 flip-flops and each of the 14 output signals also needs a macro cell. I would expect that it does not fit in the ATF1502, even if logic doubling is enabled.

hneemann, 2018-01-12:
It is possible to reduce the circuit to 18 flip-flops and then it seems to fit in the ATF1502! :-)
I really did not expect that!

[Bin2LED7.zip](https://github.com/hneemann/Digital/files/1625780/Bin2LED7.zip)

roy77, 2018-01-12:
Thank you for your work.
Yes, it was a misunderstanding. I assign the decimal point of the 7 segment to a fixed value in the truth table, so the constant was not direct obvious for me. The error message made me suspect that the obvious constants I used were the problem.

roy77, 2018-01-12:
Here the best implementation I found.
[Bin2LED7Net.zip](https://github.com/hneemann/Digital/files/1625928/Bin2LED7Net.zip)
A wiki would be nice to collect such Digital circuits.

hneemann, 2018-01-12:
Could you run your circuit on the ATF1502? 
There are some expressions containing a large number of product terms which enforces cascading of macro cells. Maybe it does not fit with the necessary pin assignment.

roy77, 2018-01-12:
I'll test it on Monday. I have an ATF1502 and an ATF1504.

hneemann, 2018-01-13:
I have clarified the error message to avoid such misunderstandings in the future.

closed 2018-01-13 by hneemann

roy77, 2018-01-15:
The last version runs, but only on a ATF1504. 
The truth table of the sequential version was to big, so Digital do not allow me to build a tt2 file.

hneemann, 2018-01-15:
Please try again the latest version.
There was a weakness in the model analyzer, which has annoyed me for a long time.
A few months ago I tried to fix it, but I did not like the solution. Last weekend I finally found a useful solution.
Now the circuits may be a bit bigger.

roy77, 2018-01-16:
Now both variants (sequential and combinatorial) work. The analysis of the sequential take about two minutes, the combinatorial only a second. The sequential was than easier to fit. The fitter stops after Filter_Pass 2. The combinatorial need a third Filter_Pass.